dusty-nv/jetson-inference #1: where to get the file "gie.aarch64-cuda7.0-1.0-ea.tar.gz"

liyouyu2008, 2016-08-15:


dusty-nv, 2016-08-15:
It will be available coming soon in JetPack, I'll post the link here.

oscarbg, 2016-08-30:
Seems GIE RC is avaiable judging from a tweet from NV Dev..

kaishijeng, 2016-09-08:
Dusty-NV

    Can you provide a lint to to get the file "gie.aarch64-cuda7.0-1.0-ea.tar.gz" ?

Thanks

dusty-nv, 2016-09-09:
It looks like this is the link for now to join RC: [https://developer.nvidia.com/gpu-inference-engine](https://developer.nvidia.com/gpu-inference-engine)

kaishijeng, 2016-09-09:
I will try to download it from the link.
Also see there is detectnet application in the jetson-inference . What framerate can you get with tx1?
I am not able to get a good framerate with yolo or ssd (4 or 5 frame/sec) object detection on tx1.

Thanks

kaishijeng, 2016-09-11:
I download and install JetPack-L4T-2.2.1-linux-x64.run to TX1, but I still can not find GIE package.
I am not able to find the download of  https://developer.nvidia.com/gpu-inference-engine
Thanks,

dusty-nv, 2016-09-11:
OK, can you contact me about the Developer site.  dustinf@nvidia.com

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

dusty-nv, 2016-09-14:
NVIDIA TensorRT (formerly known as GIE) is now publicly available as part of [JetPack 2.3](https://developer.nvidia.com/embedded/jetpack) for Jetson TX1.   See this [Parallel ForAll](https://devblogs.nvidia.com/parallelforall/jetpack-doubles-jetson-tx1-deep-learning-inference/) article for the latest performance benchmarks.   The [master](https://github.com/dusty-nv/jetson-inference) branch of this repo is updated for supporting JetPack 2.3 with Ubuntu 16.04 on TX1.

closed 2016-09-14 by dusty-nv