dusty-nv/jetson-inference #6: glibconfig.h: No such file or directory

zimenglan-sysu-512, 2016-09-20:
hi, @dusty-nv, when i make, i meet this problem, but i can't fix it, can you do me a favor?
```
/usr/include/glib-2.0/glib/gtypes.h:32:24: fatal error: glibconfig.h: No such file or directory
```
thanks.

zimenglan-sysu-512, 2016-09-20:
i chance "/usr/include/glib-2.0" to "/usr/lib/x86_64-linux-gnu/glib-2.0/include", so it can make it successfully.

closed 2016-09-20 by zimenglan-sysu-512

sonichy, 2017-02-18:
sudo cp /usr/lib/x86_64-linux-gnu/glib-2.0/include/glibconfig.h /usr/include/glib-2.0/glibconfig.h

walterzhaoJR, 2018-02-25:
gcc `pkg-config --cflags glib-2.0` test.c `pkg-config --libs glib-2.0`

saveriomiroddi, 2018-04-18:
> sudo cp /usr/lib/x86_64-linux-gnu/glib-2.0/include/glibconfig.h /usr/include/glib-2.0/glibconfig.h

this worked (thanks!), however, I think a more idiomatic approach would be to use a symlink (`sudo ln -s /usr/lib/x86_64-linux-gnu/glib-2.0/include/glibconfig.h /usr/include/glib-2.0/`).

pphhss, 2019-07-09:
> gcc `pkg-config --cflags glib-2.0` test.c `pkg-config --libs glib-2.0`
> 
> it's useful

> gcc `pkg-config --cflags glib-2.0` test.c `pkg-config --libs glib-2.0`
> 
> it's useful

Thank!!!

anselmobattisti, 2019-08-18:
At vistual studio code in the "Include path" of C compiler i needed to place the lines in this excatly oder:

`${workspaceFolder}/**
/usr/include/glib-2.0/**
/usr/include/gstreamer-1.0/**`

the /** is important to do recursive search in the folders

vlad-iancu, 2021-11-28:
> sudo cp /usr/lib/x86_64-linux-gnu/glib-2.0/include/glibconfig.h /usr/include/glib-2.0/glibconfig.h

Mine was in `/usr/lib/glib-2.0/include`

messiasv, 2023-10-24:
> > sudo cp /usr/lib/x86_64-linux-gnu/glib-2.0/include/glibconfig.h /usr/include/glib-2.0/glibconfig.h
> 
> this worked (I had the same problem on a different VS Code project), however, I think a more idiomatic approach would be to use a symlink (`sudo ln -s /usr/lib/x86_64-linux-gnu/glib-2.0/include/glibconfig.h /usr/include/glib-2.0/`).

In my case, I was using a Raspberry Pi Zero W (which has an armv6l architecture), so I used the following symlink:
`sudo ln -s /usr/lib/arm-linux-gnueabihf/glib-2.0/include/glibconfig.h /usr/include/glib-2.0/`