dusty-nv/jetson-inference #8: Using custom Digits trained model in TX1 inference.

loukbabi, 2016-09-20:
Hello Dusty.

I managed to successfully run your example with no issues (Jetpack 2.3). I would like to use my own DetectNet, which I trained using Digits 4 as shown in this example (https://github.com/NVIDIA/DIGITS/tree/master/examples/object-detection), for inference on TX1. I believe that I have changed all the corresponding fields of your repository in order to be able to import a new NetworkType and I copied my own .prototxt and .caffemodel files in the build/aarch64/bin directory. After running the imagenet-camera example I get the following error:

 ./imagenet-camera cardetect
imagenet-camera
  args (2):  0 [./imagenet-camera]  1 [cardetect]  

[gstreamer] initialized gstreamer, version 1.8.2.0
[gstreamer] gstreamer decoder pipeline string:
nvcamerasrc fpsRange="30.0 30.0" ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, format=(string)NV12 ! nvvidconv flip-method=2 ! video/x-raw ! appsink name=mysink

imagenet-camera:  successfully initialized video device
    width:  1280
   height:  720
    depth:  12 (bpp)

[GIE]  platform has FP16 support.
[GIE]  loading cardetect.prototxt bvlc_cardetect.caffemodel
[libprotobuf ERROR google/protobuf/text_format.cc:290] 
Error parsing text-format ditcaffe.NetParameter: 67:31: Message type "ditcaffe.LayerParameter" has no field named "detectnet_groundtruth_param".
Could not parse deploy file
[GIE]  failed to parse caffe network
failed to load bvlc_cardetect.caffemodel
failed to load bvlc_cardetect.caffemodel
imageNet -- failed to initialize.
imagenet-console:   failed to initialize imageNet

Tracing back this leads me to the tensorNet.h file where I found that the error occurs during the parser->parse(...) command. Is there any idea on why the detectnet_groundtruth_param field is missing from my trained model?

Thank you!

dusty-nv, 2016-09-27:
Hi loukbabi, 

GIE does not support the Python clustering layer.  It should be removed from the prototxt and re-run.  The detectNet class from the repo contains the clustering code instead.

Best regards,
Dustin

closed 2016-09-27 by dusty-nv

S4WRXTTCS, 2017-02-17:
I also ran into this issue, but even when you remove it from the prototxt it will come back with the same error message.  Apparently it got far enough to cache it, and now it pulls up the cache'd one. 

I haven't gotten it to work quite yet because I haven't figured out how to delete the cache'd version,