dusty-nv/jetson-inference #9: Trouble building NVCaffe on TX1

jetsonhacks, 2016-09-20:
When building nvcaffe on the TX1 on L4T 24.2 I ran into a  couple of issues using the instructions listed on: https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-nvcaffe.md

1. While loading dependencies, libboost-thread1.55-dev is not found. The current library on L4T 24.2 appears to be libboost-thread1.58-dev. The regular caffe branch uses libboost-all-dev, which I believe includes libboost-thread1

2.  When compiling, there is an error with the hdf5 include and library files. Error:
src/caffe/net.cpp:8:18: fatal error: hdf5.h: No such file or directory
This appears to be an Ubuntu 16.04 issue. One way to fix it is to add hdf5 to the INCLUDE_DIRS environment variable when configuring the Makefile.config. On regular caffe, this is something like:
echo "INCLUDE_DIRS += /usr/include/hdf5/serial/" >> Makefile.config
when building the Makefile. There is a similar issue with finding the actual libraries themselves. Discussion:
https://github.com/BVLC/caffe/issues/2347
One solution is to modify LIBRARY_DIRS in Makefile.config to include the hdf5 libraries as needed.

Configuration:
L4T 24.2
CUDA 8.0
gcc version 5.4.0 20160609 (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.2)

dusty-nv, 2016-10-18:
Hi Jim, here is a procedure for installing nvcaffe / pycaffe that I found worked on JetPack 2.3 / L4T R24.2:

```
Fresh Jetpack 2.3 install from NVIDIA website

sudo apt-get update -y
sudo apt-get install cmake -y

# General dependencies
sudo apt-get install libprotobuf-dev libleveldb-dev libsnappy-dev \
libhdf5-serial-dev protobuf-compiler -y
sudo apt-get install --no-install-recommends libboost-all-dev -y

# BLAS
sudo apt-get install libatlas-base-dev -y

# Remaining Dependencies
sudo apt-get install libgflags-dev libgoogle-glog-dev liblmdb-dev -y

# Python Dependencies
sudo apt-get install python-dev python-numpy python-skimage python-protobuf -y

# Clone Caffe (experimental FP16 version)
cd ~
git clone https://github.com/nvidia/caffe -b experimental/fp16
cd caffe
cp Makefile.config.example Makefile.config

# Remove cmake folder and CMakeLists files
rm -rf cmake/ CMakeLists.txt

# Edit the following lines in Makefile.config

	sed -i 's/# NATIVE_FP16/NATIVE_FP16/g' Makefile.config
	sed -i 's/# USE_CUDNN/USE_CUDNN/g' Makefile.config
	sed -i 's/-gencode arch=compute_50,code=compute_50/-gencode arch=compute_53,code=sm_53 -gencode arch=compute_53,code=compute_53/g' Makefile.config

	INCLUDE_DIRS := $(PYTHON_INCLUDE) /usr/local/include
	becomes
	INCLUDE_DIRS := $(PYTHON_INCLUDE) /usr/local/include /usr/include/hdf5/serial/

	LIBRARY_DIRS := $(PYTHON_LIB) /usr/local/lib /usr/lib
	becomes
	LIBRARY_DIRS := $(PYTHON_LIB) /usr/local/lib /usr/lib /usr/lib/aarch64-linux-gnu/hdf5/serial/

# Fix HDF5 linking issue
sudo ln -s /usr/lib/aarch64-linux-gnu/libhdf5_serial.so.10 /usr/lib/aarch64-linux-gnu/libhdf5.so
sudo ln -s /usr/lib/aarch64-linux-gnu/libhdf5_serial_hl.so.10 /usr/lib/aarch64-linux-gnu/libhdf5_hl.so

# Build caffe
make -j4
make pycaffe
make distribute

# Edit the following lines in .bashrc and then source .bashrc
export PATH=/usr/local/cuda-8.0/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/home/ubuntu/caffe/build/tools
export PYTHONPATH=/home/ubuntu/caffe/python:$PYTHONPATH

# Install ipython
sudo apt-get install ipython ipython-notebook python-pandas -y
````

jetsonhacks, 2016-10-18:
Thanks Dusty!
It's probably worth updating:

https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-nvcaffe.md

with this information.

closed 2016-10-18 by jetsonhacks