dusty-nv/jetson-inference #10: Drawing Rectangle

SeougnSeon, 2016-09-23:
I want to draw the rectangle from the results of detectNet.

I think I should use opengl.

right? Do you have plan to implement rectangle function?

dusty-nv, 2016-09-23:
Hi, yesterday I added a rectangle outline overlay function in [https://github.com/dusty-nv/jetson-inference/blob/master/cuda/cudaOverlay.h](cudaOverlay).   I also added a function to [https://github.com/dusty-nv/jetson-inference/blob/master/detectNet.h](detectNet) class which will oversee drawing the rects. 

The outline drawing is implemented in CUDA so it can also easily be used for offscreen processing, like to a file or compressed wireless video stream.

SeougnSeon, 2016-09-24:
Thank you I made a detectnet-camera demo by using overlay fucntion.

closed 2016-09-24 by SeougnSeon