dusty-nv/jetson-inference #14: How to Install on ubuntu 14.04 with custom host

FaceOP, 2016-10-03:
Hi,
I am try to install on my ubutu host.  ubuntu 14.04 and cuda version 8.0
But it seen many issue, I reference the another issue "Build problems on Ubuntu 14.04 Host" and he's CMakeLists.txt but it still has error with 
uint32_t undefined  and I put #include <cstdint> on the of cudaOverlay.cu and cudaOverlay.h
It seen ok to run.
But after it ,another question is coming.
It say Nvinfer.h : no such file or directory
How to slove it 
Thank you everyone!!

Piyush3dB, 2016-10-04:
Could you post the error as it appears in your build logs?

FaceOP, 2016-10-05:
Here is my appears 

-------------------------------------------------------------------------------------------
[  4%] Building CXX object CMakeFiles/jetson-inference.dir/imageNet.cpp.o
In file included from /home/ubuntu/Downloads/jetson-inference/imageNet.h:9:0,
                 from /home/ubuntu/Downloads/jetson-inference/imageNet.cpp:5:
/home/ubuntu/Downloads/jetson-inference/tensorNet.h:9:21: fatal error: NvInfer.h: No such file or directory
 #include "NvInfer.h"
                     ^
compilation terminated.
make[2]: *** [CMakeFiles/jetson-inference.dir/imageNet.cpp.o] Error 1
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
make: *** [all] Error 2

-------------------------------------------------------------------------------------------

dusty-nv, 2016-10-05:
Hi, if you were on Jetson, JetPack would have already installed TensorRT and the header referenced above.   However, since you are on desktop, you will need to obtain TensorRT installer through the NVIDIA [Registered Developer](https://developer.nvidia.com/tensorrt) program.

closed 2016-10-06 by dusty-nv

sulth, 2017-06-09:
$:~/jetson-inference/build$ sudo make -j8
[ 5%] Building CXX object CMakeFiles/jetson-inference.dir/imageNet.cpp.o
[ 5%] Building CXX object CMakeFiles/jetson-inference.dir/detectNet.cpp.o
[ 8%] Building CXX object CMakeFiles/jetson-inference.dir/segNet.cpp.o
[ 10%] Building CXX object CMakeFiles/jetson-inference.dir/util/camera/gstUtility.cpp.o
In file included from /home/admin/jetson-inference/detectNet.h:9:0,
from /home/admin/jetson-inference/detectNet.cpp:5:
/home/admin/jetson-inference/tensorNet.h:9:21: fatal error: NvInfer.h: No such file or directory
#include "NvInfer.h"
^
compilation terminated.
In file included from /home/admin/jetson-inference/segNet.h:9:0,
from /home/admin/jetson-inference/segNet.cpp:5:
/home/admin/jetson-inference/tensorNet.h:9:21: fatal error: NvInfer.h: No such file or directory
#include "NvInfer.h"
^
compilation terminated.
CMakeFiles/jetson-inference.dir/build.make:1373: recipe for target 'CMakeFiles/jetson-inference.dir/detectNet.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/detectNet.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /usr/include/glib-2.0/glib/galloca.h:32:0,
from /usr/include/glib-2.0/glib.h:30,
from /usr/include/gstreamer-1.0/gst/gst.h:27,
from /home/admin/jetson-inference/util/camera/gstUtility.h:9,
from /home/admin/jetson-inference/util/camera/gstUtility.cpp:5:
/usr/include/glib-2.0/glib/gtypes.h:32:24: fatal error: glibconfig.h: No such file or directory
[ 13%] Building CXX object CMakeFiles/jetson-inference.dir/tensorNet.cpp.o
compilation terminated.
CMakeFiles/jetson-inference.dir/build.make:1421: recipe for target 'CMakeFiles/jetson-inference.dir/segNet.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/segNet.cpp.o] Error 1
In file included from /home/admin/jetson-inference/imageNet.h:9:0,
from /home/admin/jetson-inference/imageNet.cpp:5:
/home/admin/jetson-inference/tensorNet.h:9:21: fatal error: NvInfer.h: No such file or directory
#include "NvInfer.h"
^
compilation terminated.
CMakeFiles/jetson-inference.dir/build.make:1517: recipe for target 'CMakeFiles/jetson-inference.dir/util/camera/gstUtility.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/util/camera/gstUtility.cpp.o] Error 1
CMakeFiles/jetson-inference.dir/build.make:1397: recipe for target 'CMakeFiles/jetson-inference.dir/imageNet.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/imageNet.cpp.o] Error 1
[ 16%] Building CXX object CMakeFiles/jetson-inference.dir/util/camera/gstCamera.cpp.o
In file included from /home/admin/jetson-inference/tensorNet.cpp:5:0:
/home/admin/jetson-inference/tensorNet.h:9:21: fatal error: NvInfer.h: No such file or directory
#include "NvInfer.h"
^
compilation terminated.
CMakeFiles/jetson-inference.dir/build.make:1445: recipe for target 'CMakeFiles/jetson-inference.dir/tensorNet.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/tensorNet.cpp.o] Error 1
In file included from /usr/include/glib-2.0/glib/galloca.h:32:0,
from /usr/include/glib-2.0/glib.h:30,
from /usr/include/gstreamer-1.0/gst/gst.h:27,
from /home/admin/jetson-inference/util/camera/gstCamera.h:8,
from /home/admin/jetson-inference/util/camera/gstCamera.cpp:5:
/usr/include/glib-2.0/glib/gtypes.h:32:24: fatal error: glibconfig.h: No such file or directory
compilation terminated.
CMakeFiles/jetson-inference.dir/build.make:1565: recipe for target 'CMakeFiles/jetson-inference.dir/util/camera/gstCamera.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/util/camera/gstCamera.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/jetson-inference.dir/all' failed
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

While iam trying to install jetson inference on host pc as mentioned in the manual,iam incurring with this error.please someone help

Shubodh, 2018-06-22:
Did you find the answer?

aikivl3, 2018-11-30:
@Shubodh have you solved your problem yet?