dusty-nv/jetson-inference #17: version of caffe being benchmarked

weishengchong, 2016-10-20:
Hi,
for Caffe mentioned in readme's chart below, may I know which Caffe branch it is referring to?

[image]

I think it is not Nvidia's experimental/fp16 [Caffe branch ](https://github.com/NVIDIA/caffe/tree/experimental/fp16 ) mentioned in [Tx1 white paper](https://www.nvidia.com/content/tegra/embedded-systems/pdf/jetson_tx1_whitepaper.pdf). 
This is because above graph shows for Googlenet batch1@fp16, above 51fps is 1.6x faster than 33fps in table below extracted from [Tx1 white paper](https://www.nvidia.com/content/tegra/embedded-systems/pdf/jetson_tx1_whitepaper.pdf) 
[image]

Thank you.

dusty-nv, 2016-10-20:
The caffe branch was experimental/fp16 and max clocks. There had been JetPack/cuDNN/OS updates since initial Jetson TX1 launch.

dusty-nv, 2016-10-20:
Also, see issue [https://github.com/dusty-nv/jetson-inference/issues/9](9) for updated nvcaffe install instructions for JetPack 2.3 / Ubuntu 16.04.  Thanks!

closed 2016-10-20 by dusty-nv