dusty-nv/jetson-inference #19: Cuda error in file src/implicit_gemm.cu at line 406: invalid device function

lymhust, 2016-10-27:
I have successfully compiled the code. However, when run the detectent-console demo, I got the following error:

```
detectnet-console
  args (2):  0 [./detectnet-console]  1 [peds-004.p]  

[GIE]  attempting to open cache file multiped-500/snapshot_iter_178000.caffemodel.tensorcache
[GIE]  cache file not found, profiling network model
[GIE]  platform does not have FP16 support.
[GIE]  loading multiped-500/deploy.prototxt multiped-500/snapshot_iter_178000.caffemodel
[GIE]  configuring CUDA engine
[GIE]  building CUDA engine
Cuda error in file src/implicit_gemm.cu at line 406: invalid device function
detectnet-console: src/implicit_gemm.cu:769: virtual int dit::ImplicitGemm::run(void*, const void*, const void*, int, cudaStream_t): Assertion `mParamsSizeInBytes<=sizeof(ImplicitGemmLargeKernelParams)' failed.
Aborted (core dumped)
```
Can anyone provide me some help about it? Thanks so much!

dusty-nv, 2016-11-01:
Hmm I am unfamiliar with the error.   Was this from a Jetson TX1 with JetPack 2.3, or another platform?

lymhust, 2016-11-02:
I have tried GIE on GTX750Ti, GTX780Ti, GTXTitanX, Quadro K1200 and DrivePX2.

This error occurs on GTX750Ti and Quadro K1200. I think the libgie1 does not support for these two platforms.

18901180429, 2016-11-03:
Correct. GIE is supported in Maxwell and Pascal platform.

closed 2017-03-23 by dusty-nv