dusty-nv/jetson-inference #20: TX1 Running the Live Camera Detection Demo FPS

FaceOP, 2016-10-31:
Hello!!!
I successful Building from Source on TX1(Before is ubuntu 14.04 + GTX980)
I try the Running the Live Camera Detection Demo
I have some question with the FPS.

./detectnet-camera multiped-500  FPS is about 6.7 FPS
./detectnet-camera ped-100 FPS is about 6.7 FPS
./detectnet-camera facenet-120 is about 12.3 FPS

Does the FPS is correct ? It looks a little slow...
How can I accelerate it ?

Another question is , I put my own model create by https://github.com/NVIDIA/DIGITS/tree/digits-4.0/examples/object-detection
And change "multiped-500 file" to it . It successful to detect car.
But the FPS still 6~7.
 How can I accelerate it ?

FaceOP, 2016-11-01:
I use the https://github.com/dusty-nv/jetson-scripts/blob/master/jetson_max_l4t.sh to max performance with tx1.
And it looks a little accelerate 6.7FPS up to 7.0FPS

forever3000, 2016-11-17:
@FaceOP ,

I tried to get source from https://github.com/ross-abaco/jetson-inference to support USB camera, the FPS I got about 8 when running this command ./detectnet-camera ped-100. I think it should be about 20fps as this graph : https://devblogs.nvidia.com/parallelforall/wp-content/uploads/2016/09/Figure_3-1.png

Anyone have the better result ?

charterscruz, 2017-01-26:
@forever3000 @FaceOP Hi,

If you look carefully at the vertical axis of the image in the previous link, you will actually see that it is "frames per second per watt". The graph is about energy efficiency and not pure performance.

TakeX1, 2017-02-06:
Hi.
I succeeded in installation of OpenFrameworks to TX1.
        [http://takesan.hatenablog.com/entry/2016/12/19/214455](url)
And detectnet(by jetson-inference) multiped-500 FPS is about 7.8~8 FPS and Camera FPS 60FPS by OF.
       [http://takesan.hatenablog.com/entry/2017/02/03/201108](url)
I'm sorry, it is all Japanese.
      By the way. 'jetson_clocks.sh'  is not effective by this 'jetson-inference'.  Why?

closed 2017-03-23 by dusty-nv

TakeX1, 2017-03-24:
Hi Dustin .

What do you mean?