dusty-nv/jetson-inference #22: Inferencing on desktop PASCAL GPU with TensorRT 1.0 RC on CUDA 8.0

xhuvom, 2016-11-10:
I have recently obtained and installed TensorRT 1.0RC on an early access program. However, this jetson-inference repo should work on PASCAL based GTX 1080 running CUDA 8.0 with the GIE provided by NVIDIA for demo. Running 'make' command generates error: 

`Error limit reached.
100 errors detected in the compilation of "/tmp/tmpxft_000009d1_00000000-7_cudaOverlay.cpp1.ii".
Compilation terminated.
CMake Error at jetson-inference_generated_cudaOverlay.cu.o.cmake:264 (message):
  Error generating file
  /home/xhuv/jetson-inference/build/CMakeFiles/jetson-inference.dir/cuda/./jetson-inference_generated_cudaOverlay.cu.o
`

What change should be made to run this code on desktop GPUs other than TX1?

karstenBehrendt, 2016-11-15:
Adding a little to this issue. Does jetson_inference support x86_86?
On master, I get a similar error as @xhuvom:
```
~/repos/jetson-inference/build$ make
[  4%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/cuda/./jetson-inference_generated_cudaFont.cu.o
<command-line>:0:7: warning: ISO C99 requires whitespace after the macro name [enabled by default]
<command-line>:0:7: warning: ISO C99 requires whitespace after the macro name [enabled by default]
<command-line>:0:7: warning: ISO C99 requires whitespace after the macro name [enabled by default]
/usr/lib/gcc/x86_64-linux-gnu/4.8/include/stddef.h(432): error: identifier "nullptr" is undefined

/usr/lib/gcc/x86_64-linux-gnu/4.8/include/stddef.h(432): error: expected a ";"

/usr/include/x86_64-linux-gnu/c++/4.8/bits/c++config.h(190): error: expected a ";"
....
Error limit reached.
100 errors detected in the compilation of "/tmp/tmpxft_00004501_00000000-7_cudaFont.cpp1.ii"
```
When trying L4T-R24.1, I get:
```
[  5%] Building CXX object CMakeFiles/jetson-inference.dir/imageNet.cpp.o
In file included from ~/repos/jetson-inference-L4T-R24.1/imageNet.h:9:0,
                 from ~/repos/jetson-inference-L4T-R24.1/imageNet.cpp:5:
~/repos/jetson-inference-L4T-R24.1/caffeToGIE.h:9:19: fatal error: Infer.h: No such file or directory
 #include "Infer.h"
```

I am running Ubuntu 14.04, TensorRT 1.0RC, Cuda 8, nvidia 370 with a Titan X Pascal.
I understand that it currently is in beta and that there may be minor flaws / missing support.
Thanks!

dusty-nv, 2016-12-01:
Hi, can you pull master again and add `-std=c++11"` to the `CUDA_NVCC_FLAGS` in CMakeLists?   It looks like your error is from not having c++11 support enabled while compiling a .cu file.

closed 2017-03-23 by dusty-nv

bpinaya, 2017-03-30:
I've added the flag and when trying on a different setup it outputs:
`[  3%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/util/cuda/./jetson-inference_generated_cudaFont.cu.o
<command-line>:0:7: warning: ISO C99 requires whitespace after the macro name [enabled by default]
<command-line>:0:7: warning: ISO C99 requires whitespace after the macro name [enabled by default]
<command-line>:0:7: warning: ISO C99 requires whitespace after the macro name [enabled by default]
<command-line>:0:7: warning: ISO C99 requires whitespace after the macro name [enabled by default]
/usr/include/c++/4.8/type_traits(269): error: identifier "__float128" is undefined

1 error detected in the compilation of "/tmp/tmpxft_00005fcf_00000000-9_cudaFont.compute_62.cpp1.ii".
`
But that seems to be a boost 1.6 issue
https://svn.boost.org/trac/boost/ticket/11852
Will try latter.