dusty-nv/jetson-inference #24: I want to add my customized layer

SeougnSeon, 2016-12-10:
I want to add my customized layer in Jetson-Inference demo.

Currently, GIE does not support customized layers as well as latest caffe versions.

I want to add my caffe distribute(libcaffe.so) in this camera demo by removing GIE.

Is it possible?

suhyung, 2017-01-19:
I also strongly hope this feature

e2sh83, 2017-01-26:
Can anybody answer for this request? 
It would be great if applying customized layer is possible.

dusty-nv, 2017-01-26:
Hi there, custom layers are a feature of TensorRT2, the next version.

For now, to add support for custom layers, please use Caffe or your own code.  The Tegra Multimedia API which JetPack installs, includes a sample of using TensorRT and cuDNN in tandem to implement a 'custom' function.

For example, the detectNet and segNet example networks that are included in this repo, I removed superfluous layers to get it working.  Typically these are at the end of the pipeline and are linear operators (like cropping, ect.) as opposed to having non-zero learning rate.

SeougnSeon, 2017-01-31:
Thank you for your kind explanation.

closed 2017-01-31 by SeougnSeon