dusty-nv/jetson-inference #25: Build fails

DVEfremov, 2016-12-14:
**Steps to reproduce** 

* Ubuntu 16.x
* git clone https://github.com/dusty-nv/jetson-inference
* mkdir build
* cd build
* cmake ../
* make

```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_CUDART_LIBRARY (ADVANCED)
    linked by target "jetson-inference" in directory /home/daniil/Projects/Dividiti/tensort/jetson-inference
    linked by target "imagenet-console" in directory /home/daniil/Projects/Dividiti/tensort/jetson-inference/imagenet-console
    linked by target "imagenet-camera" in directory /home/daniil/Projects/Dividiti/tensort/jetson-inference/imagenet-camera
    linked by target "detectnet-console" in directory /home/daniil/Projects/Dividiti/tensort/jetson-inference/detectnet-console
    linked by target "detectnet-camera" in directory /home/daniil/Projects/Dividiti/tensort/jetson-inference/detectnet-camera
    linked by target "segnet-console" in directory /home/daniil/Projects/Dividiti/tensort/jetson-inference/segnet-console
CUDA_TOOLKIT_INCLUDE (ADVANCED)
   used as include directory in directory /home/daniil/Projects/Dividiti/tensort/jetson-inference
...
-- Configuring incomplete, errors occurred!
See also "/home/daniil/Projects/Dividiti/tensort/jetson-inference/build/CMakeFiles/CMakeOutput.log".
See also "/home/daniil/Projects/Dividiti/tensort/jetson-inference/build/CMakeFiles/CMakeError.log".
make: *** No targets specified and no makefile found.  Stop.

```

DVEfremov, 2016-12-14:
Builds fine at 
Linux tegra-ubuntu 3.10.96-tegra #1 SMP PREEMPT Wed Nov 9 19:42:57 PST 2016 aarch64 aarch64 aarch64 GNU/Linux

Looks my ubuntu env problem.

closed 2016-12-14 by DVEfremov

funge, 2017-05-06:
I have the same issue, how did you fix? Thanks

kalpitsmehta, 2018-01-20:
Did anyone find a solution to this?

winecos, 2020-03-29:
solved it to this ? @DVEfremov @funge  
thx