dusty-nv/jetson-inference #27: Detectnet fine tuning

VanitarNordic, 2017-01-09:
Hello Sir,

First I appreciate about your provided information.
My question is basically about DetectNet. 

1) You have provided 3 trained models. may I ask you where can I download them? I mean ped-100, multiped-500, facenet-120.

2) Do you have any more trained model rather than above mentioned ones?

3) I use DIGITS 5.1, and I decide to find tune model instead of training from scratch. Then may I ask you which `_.prototxt` should be used in accompany with CaffeModel? I mean `deploy.prototxt` or `original.prototxt` or `solver.prototxt` or `train_val.prototxt`?

VanitarNordic, 2017-01-09:
Okay, I solved the first question, but please help me with second and third.

dusty-nv, 2017-01-10:
Hi, you can find the original DetectNet and training guide in included with DIGITS here:.  https://github.com/NVIDIA/DIGITS/tree/master/examples/object-detection

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

VanitarNordic, 2017-01-10:
Hello,

Yes, I know about that but I feel they have not revealed everything. if the intention is to fine tune the model, some layers must be renamed but they did NOT.

closed 2017-03-23 by dusty-nv