dusty-nv/jetson-inference #31: jetson-inference detectnet-camera can not using training model?

BrainyBeeLy, 2017-01-21:
Dgits training model success.
https://github.com/dusty-nv/jetson-inference/tree/master/detectnet-camera
jetson-inference  detectnet-camera can not using this model, error follow: 

[GIE] attempting to open cache file vehicle/snapshot_iter_21120.caffemodel.tensorcache
[GIE] cache file not found, profiling network model
[GIE] platform has FP16 support.
[GIE] loading vehicle/deploy.prototxt vehicle/snapshot_iter_21120.caffemodel
**[GIE] failed to retrieve tensor for output 'coverage'
[GIE] failed to retrieve tensor for output 'bboxes'**
[GIE] configuring CUDA engine
[GIE] building CUDA engine

dusty-nv, 2017-01-21:
Hi there, can you post the deploy.prototxt?   Thank you.

https://github.com/dusty-nv/jetson-inference/tree/master/detectnet-camera

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

BrainyBeeLy, 2017-01-22:
[deploy.prototxt](E:\saveStorage\20170121-172331-8693_epoch_30.0)

BrainyBeeLy, 2017-01-22:
> 
[deploy.txt](https://github.com/dusty-nv/jetson-inference/files/721598/deploy.txt)

closed 2017-03-23 by dusty-nv

adithya-p, 2018-02-26:
@BrainyBeeLy @dusty-nv @lukeyeager

Did you find a way to fix this? Can you please help me with the same? This issue has been closed even though there are no answers. No answers, but duplicate questions and the comment section ends in a loop in which one points to the other.

scottire, 2018-07-23:
@BrainyBeeLy @dusty-nv @lukeyeager @adithya-p
Has anyone found a solution to this? 
I am having the same issue. The lack of error message is confusing, it has completely stalled with no indication of why. 
Here is my [deploy.prototxt](https://github.com/dusty-nv/jetson-inference/files/2219530/deploy.txt)