dusty-nv/jetson-inference #32: webcam usage

ShervinAr, 2017-01-24:
Dear all,
Thanks for the codes and comments provided.
I have been able to compile and run the examples corresponding to console (detectnet-console and imagenet-console) on a "PC" running Ubuntu 14.04 but unfortunately the webcam cannot be used (the detectnet-camera and imagenet-camera apps do not run). So far I have tested two different webcams and neither one could be used for detection/recognition purposes. Running imagenet-camera/detectnet-camera apps I get some run-time errors such as 'cannot convert rgbtorgba'/'could not capture frame'...
The webcams can be initialized but I get some black screen with some noise and not a proper image of a scene..
any comments or help is very much appreciated

dusty-nv, 2017-01-24:
Hi ShervinAr, have you compiled the camera programs with #define DEFAULT_CAMERA 0 near the top?  This will trigger V4L2 getting used.  What colorspace(s) do your webcams support?  If you aren't sure, v4l2-ctl utility should enumerate them.

Thanks,
Dustin

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

ShervinAr, 2017-01-24:
@dusty-nv  Hi Dustin, Thanks for your quick response.
Yes I have used #define DEFAULT_CAMERA set to 0 near the top.
One of my webcams supports RBG colorspace.
For which I got the errors mentioned earlier.

many thanks,
Shervin

ShervinAr, 2017-01-28:
@dusty-nv  running v4l2-ctl -V gives: 

"Format Video Capture:
	Width/Height  : 640/480
	Pixel Format  : 'YUYV'
	Field         : None
	Bytes per Line: 1280
	Size Image    : 614400
	Colorspace    : SRGB
	Custom Info   : feedcafe
"
I got the error:

"gstbasesrc.c:2865  gst_base_src_loop
              error: Internal data flow error.
[gstreamer] GST_LEVEL_WARNING GstV4l2Src basesrc
              gstbasesrc.c:2865  gst_base_src_loop
              error: streaming task paused, reason not-negotiated (-4)
[gstreamer] gstreamer decoder onEOS
[gstreamer] gstreamer v4l2src0 ERROR Internal data flow error.
[gstreamer] gstreamer Debugging info: gstbasesrc.c(2865): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming task paused, reason not-negotiated (-4)
[gstreamer] gstreamer changed state from READY to PAUSED ==> mysink

imagenet-camera:  camera open for streaming

imagenet-camera:  failed to capture frame
imagenet-camera:  failed to convert from NV12 to RGBA
imageNet::Classify( 0x(nil), 1280, 720 ) -> invalid parameters
[cuda]   cudaNormalizeRGBA((float4*)imgRGBA, make_float2(0.0f, 255.0f), (float4*)imgRGBA, make_float2(0.0f, 1.0f), camera->GetWidth(), camera->GetHeight())
[cuda]      invalid device pointer (error 17) (hex 0x11)
[cuda]      /home/shervin/runn/jetson-inference/imagenet-camera/imagenet-camera.cpp:181
[cuda]   registered 14745600 byte openGL texture for interop access (1280x720)
"

many thanks for your help on this.

closed 2017-03-23 by dusty-nv

abdo-abaco, 2017-03-29:
Hi,

I get similar errors for the c920 webcam.
deep-learner@Deep-learner:~/jetson-inference/build/x86_64/bin$ v4l2-ctl -V
Format Video Capture:
	Width/Height  : 1280/720
	Pixel Format  : 'MJPG'
	Field         : None
	Bytes per Line: 0
	Size Image    : 1843200
	Colorspace    : SRGB
	Custom Info   : feedcafe

Upon running the imagenet-camera application the C920 lights up, the window reads 'NVIDIA Jetson TX1 | L4T R24.1 aarch64 | Ubuntu 14.04 although this was compiled for x86_64 .

These are my errors in repeated cycles:
[cuda]   cudaGetLastError()
[cuda]      invalid device function (error 8) (hex 0x08)
[cuda]      /home/deep-learner/jetson-inference/util/cuda/cudaRGB.cu:44
[cuda]   cudaRGBToRGBAf((uchar3*)input, (float4*)mRGBA[mLatestRGBA], mWidth, mHeight)
[cuda]      invalid device function (error 8) (hex 0x08)
[cuda]      /home/deep-learner/jetson-inference/util/camera/gstCamera.cpp:90
imagenet-camera:  failed to convert from NV12 to RGBA
imageNet::Classify( 0x(nil), 1280, 720 ) -> invalid parameters
[cuda]   cudaNormalizeRGBA((float4*)imgRGBA, make_float2(0.0f, 255.0f), (float4*)imgRGBA, make_float2(0.0f, 1.0f), camera->GetWidth(), camera->GetHeight())
[cuda]      invalid device pointer (error 17) (hex 0x11)
[cuda]      /home/deep-learner/jetson-inference/imagenet-camera/imagenet-camera.cpp:181
[cuda]   registered 14745600 byte openGL texture for interop access (1280x720)

Was this issue ever resolved?

Thank you,
Abdo

ShervinAr, 2017-03-29:
@abdo-abaco  check https://github.com/Abaco-Systems/jetson-inference-gv and see whether it works for you if you have not tried it yet

abdo-abaco, 2017-03-30:
With that I get:

[cuda]   cudaGetLastError()
[cuda]      invalid device function (error 8) (hex 0x08)
[cuda]      /home/deep-learner/jetson-inference-gv/cuda/cudaRGB-NV12.cu:42
[cuda]   cudaRGBToRGBAf((uint8_t*)input, (float4*)mRGBA, mWidth, mHeight)
[cuda]      invalid device function (error 8) (hex 0x08)
[cuda]      /home/deep-learner/jetson-inference-gv/camera/camera.cpp:118
[cuda]   camera -- conversion cudaRGBToRGBAf failed (1280x720)
imagenet-camera:  failed to convert from RGB to RGBAf
imageNet::Classify( 0x(nil), 1280, 720 ) -> invalid parameters
imagenet-camera:  classify failure, aborting

Thank you, will let you know if this gets resolved.

abdo-abaco, 2017-04-20:
@ShervinAr 

Found out the problem with running this demo on host. Run your deviceQuery example and find out your compute architecture. In my case 5.0 and 6.3. As mentioned make to #define DEFAULT_CAMERA  to 0 

Go to CMakeLists.txt and add the lines underneath: -gencode arch=compute_50,code=sm_50
	-gencode arch=compute_50,code=sm_50
	-gencode arch=compute_61,code=sm_61