dusty-nv/jetson-inference #33: Batch Processing

mescarra, 2017-01-31:
Just a follow-up on support for batch processing discussed in this topic:
https://devtalk.nvidia.com/default/topic/969035/jetson-tx1/embedded-deep-learning-with-jetson-mdash-nvidia-webinar/post/5060249/#5060249

nPellejero, 2017-02-03:
Hi, I met the same issue. It would be of great help to have some insight on this. Thank you in advance.

dusty-nv, 2017-02-06:
Hi there, which imaging primitive would you prefer be outfitted with batching first?   imageNet, detectNet, or segNet?

mescarra, 2017-02-06:
Hello Dustin!

imageNet would be the best choice, since what we want is to classify small patches taken from a regular grid out of a big picture, so it would be useful to have an example of the classification task.

Thanks in advance

dusty-nv, 2017-02-06:
Thanks for explaining your use-case, that sounds useful.  I will add it as soon as I can - my wife is due with twins any minute now...  And eventually I will add it to all three imaging primitives.

mescarra, 2017-02-06:
Wow! Congratulations Dustin! Big day today hehe
Thank you very much for you help!

corti-forkworks, 2017-03-01:
Hello, any news about this new feature? It would be very helpful for me as well. Thank you very much!!

dkopljar, 2017-09-28:
@dusty-nv Any progress on this batch processing feature?

rockkingjy, 2017-12-15:
@dusty-nv I need this too, exactly the same as mescarra said, any news? Or point out a way to do this, I will do it by myself if it's possible.

rockkingjy, 2017-12-15:
@mescarra @corti-forkworks @dkopljar I've done this. Thanks. Check link: https://github.com/rockkingjy/jetson-inference-batch

romzie, 2018-03-28:
@dusty-nv Any update on an official feature?

BAILOOL, 2018-08-07:
@dusty-nv @romzie any update?

closed 2023-04-27 by dusty-nv