dusty-nv/jetson-inference #35: make Error occurred：

molyswu, 2017-02-10:
/jetson-inference/build# make
[  2%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/cuda/jetson-inference_generated_cudaRGB.cu.o
<command-line>:0:2: warning: ISO C99 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C99 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C99 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C99 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C99 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C99 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C99 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C99 requires whitespace after the macro name
/usr/include/string.h: In function ‘void* __mempcpy_inline(void*, const void*, size_t)’:
/usr/include/string.h:652:42: error: ‘memcpy’ was not declared in this scope
   return (char *) memcpy (__dest, __src, __n) + __n;
                                          ^
CMake Error at jetson-inference_generated_cudaRGB.cu.o.cmake:266 (message):
  Error generating file
  /home/molys/Download_file/jetson-inference/build/CMakeFiles/jetson-inference.dir/cuda/./jetson-inference_generated_cudaRGB.cu.o

CMakeFiles/jetson-inference.dir/build.make:249: recipe for target 'CMakeFiles/jetson-inference.dir/cuda/jetson-inference_generated_cudaRGB.cu.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/cuda/jetson-inference_generated_cudaRGB.cu.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/jetson-inference.dir/all' failed
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

my machine is GTX1080+CUDA8.0+cudnn7.5
Thank you!

Jia-HongHenryLee, 2017-02-22:
Do you solve this problem? I have the same problem.

curiouser001, 2017-04-09:
Yes. I have same problem. Do you solve this problem?

Jia-HongHenryLee, 2017-04-10:
@curiouser001  Yes, I solve this problem. I just remove it and download this project's github using sudo command.
solution: 
rm -r jetson-inference
sudo git clone http://github.org/dusty-nv/jetson-inference
cd jetson-inference
mkdir build
cd build
sudo cmake ..
sudo make -j8

curiouser001, 2017-04-10:
Thank you for your answer. But these did not work. I deleted Ubuntu 14.04. I installed Ubuntu 16.04 instead. The codes are compiled. But when I run the examples, it fails like this:

`/Desktop/jetson/jetson-inference/build/x86_64/bin$ sudo ./imagenet-console orange_0.jpg output1.jpg
imagenet-console
args (3): 0 [./imagenet-console] 1 [orange_0.jpg] 2 [output1.jpg] 

[GIE] attempting to open cache file networks/bvlc_googlenet.caffemodel.2.tensorcache
[GIE] loading network profile from cache... networks/bvlc_googlenet.caffemodel.2.tensorcache
[GIE] platform does not have FP16 support.
[GIE] networks/bvlc_googlenet.caffemodel loaded
[GIE] CUDA engine context initialized with 2 bindings
[GIE] networks/bvlc_googlenet.caffemodel input binding index: 0
[GIE] networks/bvlc_googlenet.caffemodel input dims (b=2 c=3 h=224 w=224) size=1204224
[cuda] cudaAllocMapped 1204224 bytes, CPU 0x1020d000000 GPU 0x1020d000000
[GIE] networks/bvlc_googlenet.caffemodel output 0 prob binding index: 1
[GIE] networks/bvlc_googlenet.caffemodel output 0 prob dims (b=2 c=1000 h=1 w=1) size=8000
[cuda] cudaAllocMapped 8000 bytes, CPU 0x1020d200000 GPU 0x1020d200000
networks/bvlc_googlenet.caffemodel initialized.
[GIE] googlenet loaded
imageNet -- loaded 1000 class info entries
googlenet initialized.
loaded image orange_0.jpg (1920 x 1920) 58982400 bytes
[cuda] cudaAllocMapped 58982400 bytes, CPU 0x1020d400000 GPU 0x1020d400000
[cuda] cudaGetLastError()
[cuda] invalid device function (error 8) (hex 0x08)
[cuda] /home/monster/Desktop/jetson/jetson-inference/imageNet.cu:97
[cuda] cudaPreImageNetMean((float4*)rgba, width, height, mInputCUDA, mWidth, mHeight, make_float3(104.0069879317889f, 116.66876761696767f, 122.6789143406786f))
[cuda] invalid device function (error 8) (hex 0x08)
[cuda] /home/monster/Desktop/jetson/jetson-inference/imageNet.cpp:184
imageNet::Classify() -- cudaPreImageNetMean failed
imagenet-console: failed to classify 'orange_0.jpg' (result=-1)

shutting down...`

Jia-HongHenryLee, 2017-04-10:
@curiouser001  i build this code in jetson TX1 platform. the structure of jetson TX1 platform is aarch64 (arm 64bit). I think this code cannot run in x86_64 structure.

curiouser001, 2017-04-10:
@Jia-HongHenryLee But the main page says that: "Note that TensorRT samples from the repo are intended for deployment on embedded Jetson TX1 / TX2 module, however when cuDNN and TensorRT have been installed on the host side, "

dusty-nv, 2017-04-10:
The error:

[cuda] invalid device function (error 8) (hex 0x08)

Is from the CUDA kernels being compiled with compute_53/sm_53 and compute_62/sm_62 for Jetson TX1 and TX2, respectively.

Which GPU are you using?   Check it’s device capability with CUDA deviceQuery tool.  Then append you sm to jetson-inference/CMakeLists.txt line 37:

set(

        CUDA_NVCC_FLAGS

        ${CUDA_NVCC_FLAGS};

    -O3

        -gencode arch=compute_53,code=sm_53

        -gencode arch=compute_62,code=sm_62

)

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

curiouser001, 2017-04-10:
Thank you so much. I have a GTX1060 card. -Gencode belt = compute_61, code = sm_61 worked in the summer.

But real-time examples do not work.

`/jetson/jetson-inference/build/x86_64/bin$ sudo ./gst-camera 
gst-camera
  args (1):  0 [./gst-camera]  
[gstreamer] initialized gstreamer, version 1.2.4.0
[gstreamer] gstreamer decoder pipeline string:
nvcamerasrc fpsRange="30.0 30.0" ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, format=(string)NV12 ! nvvidconv flip-method=2 ! video/x-raw ! appsink name=mysink
[gstreamer] GST_LEVEL_ERROR     GST_PIPELINE
              ./grammar.y:672  priv_gst_parse_yyparse
              no element "nvcamerasrc"
[gstreamer] GST_LEVEL_ERROR     GST_PIPELINE
              ./grammar.y:942  priv_gst_parse_launch
              Unrecoverable syntax error while parsing pipeline nvcamerasrc fpsRange="30.0 30.0" ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, format=(string)NV12 ! nvvidconv flip-method=2 ! video/x-raw ! appsink name=mysink
[gstreamer] gstreamer decoder failed to create pipeline
[gstreamer]    (no element "nvcamerasrc")
[gstreamer] failed to init gstCamera

gst-camera:  failed to initialize video device
`

dusty-nv, 2017-04-10:
That’s because by default the samples are compiled to use gstreamer and nvcamerasrc which is only available on Jetson.

If you change #define DEFAULT_CAMERA -1 at the top of the file to #define DEFAULT_CAMERA 0 , instead it will use V4L2 camera #0.

curiouser001, 2017-04-10:
I've been dealing with it since yesterday. I could not do it.
Thank you very much. The program worked very well. Have a good day.

trohit920, 2018-07-30:
Hello, @curiouser001 @dusty-nv I am trying to run Jetson-Inference on a host machine and got stuck with this error. I have installed TensorRT and GStreamer and all other Cuda related requirements. I am having GTX 1080 GPU installed with CUDA 9. I have done all the steps mentioned in this post like changing -gencode arch in CMakeList.txt but still getting error.Please can you see what i am doing wrong here?

`~/jetson-inference/build$ sudo make -j8
[ 10%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/util/cuda/jetson-inference_generated_cudaRGB.cu.o
[ 10%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/util/cuda/jetson-inference_generated_cudaOverlay.cu.o
[ 10%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/jetson-inference_generated_imageNet.cu.o
[ 10%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/util/cuda/jetson-inference_generated_cudaYUV-NV12.cu.o
[ 12%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/util/cuda/jetson-inference_generated_cudaYUV-YV12.cu.o
[ 15%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/util/cuda/jetson-inference_generated_cudaNormalize.cu.o
[ 17%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/util/cuda/jetson-inference_generated_cudaFont.cu.o
[ 20%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/util/cuda/jetson-inference_generated_cudaResize.cu.o
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
[... cut ...]
/home/rohit/jetson-inference/tensorNet.cpp:413:3: error: ‘mOutputs’ was not declared in this scope
   mOutputs.push_back(l);
   ^
CMakeFiles/jetson-inference.dir/build.make:125: recipe for target 'CMakeFiles/jetson-inference.dir/tensorNet.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/tensorNet.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/jetson-inference.dir/all' failed
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
`
Waiting for your response? Thanks in advance.