dusty-nv/jetson-inference #36: How to deploy the trained model in jetson-inference code

westqnp, 2017-02-13:
Hi there...
Thanks for the code.
 I m new to training and inference. I trained a model with my own data set(nearly 300 images) in DIGITS and now trying to deploy it using Jetson-inference. I have a Jetson TX1 development kit and successfully installed the jetson-inference in it. But when I modified the caffe.model_file with caffemodel file  Im getting the following errors.
[GIE] attempting to open cache file snapshot_iter_180.caffemodel.tensorcache
[GIE] cache file not found, profiling network model
[GIE] platform has FP16 support.
[GIE] loading deploy.prototxt snapshot_iter_180.caffemodel
[GIE] failed to retrieve tensor for output 'prob'
[GIE] configuring CUDA engine
[GIE] building CUDA engine

It will be appreciated if you can give me some advice.

raaka1, 2017-02-13:
Hope this helps you
https://github.com/dividiti/ck-tensorrt/issues/7 :)

dusty-nv, 2017-02-13:
Hi westqnp, does the prototxt of your detectNet include the 'prob' output layer?  If not, you can use the detectNet class API to set a different name for the output probability layer.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

westqnp, 2017-02-13:
Thank you so much for the reply. I used imageNet_camera for image classfication. but I am not sure how to modify ilsvrc12_synset_words.txt with my data set image names ( or indexing) and plug in in the code. When I use my own caffemodel results are showing the image names from ilsvrc12_synset_words.txt for imageNet_camera.

westqnp, 2017-02-13:
Thank you so much for the reply. I used imageNet_camera for image classfication. but I am not sure how to modify ilsvrc12_synset_words.txt with my data set image names ( or indexing) and plug in in the code. When I use my own caffemodel results are showing the image names fromilsvrc12_synset_words.txt

Jia-HongHenryLee, 2017-02-23:
Hi @westqnp 
Do you have this error when you type 'make'? #41

dusty-nv, 2017-03-23:
Hi, you may need to run ```cmake ../``` again after editing to get ```ilsvcr12_synset_words.txt``` copied into the bin directory.

closed 2017-03-23 by dusty-nv