dusty-nv/jetson-inference #38: Profling leads to a FPS drop

kevinkit, 2017-02-16:
Thank you for the example code.

When I change the profiling to "true" in the TensorNet-Class, the profiling works like a charm. However, the FrameRate drops from 30 to 23 when using the GoogleNet example. 
Is there a way to speed this up? Maybe by writing the output from the profling into a file instead of printing it out or even make a second thread that runs, and prints the profling ?

I am using Nvidia Jetson TX1 Board with L4T 24.2 

Greetings

dusty-nv, 2017-02-16:
Hi Kevin, I am out of the office currently, but thanks for reporting this interesting behavior.  I had not noticed previously because I primarily use profiling mode in the console applications.  Could you try uncommenting the printf statements in the tensorNet profiling callback functions, and see if the overall Fps changes?  Thanks!

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

dusty-nv, 2017-02-16:
Typo, I meant commenting out the printf inside the callback, not uncommenting.

On Feb 16, 2017 11:27 AM, Dustin Franklin <dustinf@nvidia.com> wrote:

Hi Kevin, I am out of the office currently, but thanks for reporting this interesting behavior.  I had not noticed previously because I primarily use profiling mode in the console applications.  Could you try uncommenting the printf statements in the tensorNet profiling callback functions, and see if the overall Fps changes?  Thanks!

kevinkit, 2017-02-17:
Hello,

I commented out the line `printf(LOG_GIE "layer %s - %f ms\n", layerName, ms);` in line 114 at [tensorNet.h](https://github.com/dusty-nv/jetson-inference/blob/master/tensorNet.h) , this lead to a normal 29-30 FPS. 

If I use a simple file pointer and write the output in to a file the framerote drops to 23 if there are a lot of movements, but can get up to 27 max if there is no movement captured.

However redirecting the ouptput like (of course with the output from profiling enabled again): 

`./imagenet-camera > someFile.txt `

leads to a framerate around 30 FPS. 

On a side note, I also found out that if you increase the Batch-Size that the framerate drops, too. Should this be in a next issue?

closed 2023-04-27 by dusty-nv