dusty-nv/jetson-inference #39: Changing the Batch-Size does not show effect when it is not in the TensorNet.cpp

kevinkit, 2017-02-17:
Hello, 

When changing:

`MAX_BATCH_SIZE `

in any of the examples it will not have an effect, it has to be changed in the TensorNet.cpp to have an effect. 

Greetings

dusty-nv, 2017-03-23:
Thank you, I have modified the TensorNet class and vision primitive APIs to accept the ```MAX_BATCH_SIZE``` as an argument, so it should take effect more easily now.

closed 2017-03-23 by dusty-nv

sulthanashafi, 2017-06-27:
Changing the batch size in tensor net.h alone doesnt worked.Please may i know the locations to  change batch size.

kevinkit, 2017-06-27:
Like written above in TensorNet.cpp

sulthanashafi, 2017-06-27:
I have changed in tensorNet.h  --> LINE:bool LoadNetwork( const char* prototxt, const char* model, const char* mean=NULL,
                                      const char* input_blob="data", const char* output_blob="prob",
                                          uint32_t maxBatchSize=1 );.Please point me to the locations if iam wrong.Also,i have recompiled.But,i think few more locations are yet to be changed.Please help me.

kevinkit, 2017-06-27:
Sry I was mistaken, that was on an older version.

Anyhow, I think this should do the trick since you are changing the default parameter of the create function. Anytime the create function is called without specifying this statement the default is used. If you wish to specify an other Batchsize you must provide it. 

e.g. line 33 of imagenet-console.cpp

	// create imageNet
	imageNet* net = imageNet::Create(argc, argv);

You could change to 
`
imageNet* net = imageNet::Create(argc, argv,maxBatchSize=4);`

Anyhow remember, that the batchsizes always must be dividable by 2. So setting it to "1" will not make any change

sulthanashafi, 2017-06-27:
Thankyou.I tried it but haven't succeeded.error: 'maxBatchSize' was not declared in this scope
  detectNet* net = detectNet::Create(argc, argv,maxBatchSize=1);
                                                ^

kevinkit, 2017-06-27:
I am not quite sure if it possible in this C++ version to write it like this. maybe you have to write evey function argument, till you come to the maxbatchsize argument. 

e.g.:
`detectNet* net = detectNet::Create(argc,argv,maxBatchSize=1,NULL,NULL,IMAGENET_DEFAULT_INPUT,IMAGENET_DEFAULT_OUTPUT,4);
`
I am not sure if it will work with the NULL operators, though. 

P.S.: Please use the formatting tools for code when you submit code, to make it easier to read

sulthanashafi, 2017-06-27:
Yes i got it changed when i modified it in detectNet.h.Thankyou for your effort and time.