dusty-nv/jetson-inference #40: Feature Request - Saving Image + Label file within DetectNet-Camera.cpp

S4WRXTTCS, 2017-02-21:
I think it would be really useful for certain use cases to be able to save an image plus the label file on command.

That way we can build a dataset that can then be adjusted/fixed before adding it back to the original dataset used to create the model.  To retrain the model with the more data. 

In my case I'm actually using the object detection boundary boxes to do a separate classification network.  For example if I had hundreds of toy cars I would want to first find the cars using detectnet, and then I'd want to run a separate classification on the cars to determine the exact type. Like a Porsche 911 or a Ford F150. 

I'm trying to do this myself, but of course I immediately ran into an issue due to my lack of knowledge with CUDA (never used it before). Where I'm confused about the GPU memory space versus the CPU memory space. I'm using the saveImageRGBA function that's used within detectnet-console.cpp, but it looks like its meant for an data within the GPU memory. For detectnet-camera it looks like the frame gets captured to shared image space, and then it gets converted to RGBA within the GPU space.  I get a segmentation fault when I try using the saveImageRGBA function.

apekshaK, 2017-09-08:
Were you able to achieve this?
I am trying to do the same. I guess the imgRGBA needs to be copied to CPU memory using cudaMemcpyDeviceToHost, so as to be able to use the saveImageRGBA function.

dusty-nv, 2017-09-09:
If you make the change `camera->ConvertRGBA(imgCUDA, &imgRGBA, true)`, then imgRGBA will be zeroCopy shared memory that you can access from CPU without needing `cudaMemcpy`.  See here in the header:

[https://github.com/dusty-nv/jetson-inference/blob/f41f1c06f8fd31e17069d69f5822e0864026ac2f/util/camera/gstCamera.h#L58](https://github.com/dusty-nv/jetson-inference/blob/f41f1c06f8fd31e17069d69f5822e0864026ac2f/util/camera/gstCamera.h#L58)

closed 2023-04-27 by dusty-nv