dusty-nv/jetson-inference #41: Make Error: Building NVCC (Device) object CMakeFiles/jetson-inference.dir/cuda/jetson-inference_generated_cudaYUV-YV12.cu.o

Jia-HongHenryLee, 2017-02-23:
Can everyone help me how to solve this following error when i type 'make'?

 [  2%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/cuda/jetson-inference_generated_cudaYUV-YV12.cu.o
nvcc fatal   : Cannot compile in the 32-bit mode when the host compiler targets aarch64.
CMake Error at jetson-inference_generated_cudaYUV-YV12.cu.o.cmake:207 (message):
  Error generating
  /media/share33/User/henrylee/tx1/jetson-inference/build/CMakeFiles/jetson-inference.dir/cuda/./jetson-inference_generated_cudaYUV-YV12.cu.o

CMakeFiles/jetson-inference.dir/build.make:119: recipe for target 'CMakeFiles/jetson-inference.dir/cuda/jetson-inference_generated_cudaYUV-YV12.cu.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/cuda/jetson-inference_generated_cudaYUV-YV12.cu.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/jetson-inference.dir/all' failed
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

Thank you very much.

Jia-HongHenryLee, 2017-02-23:
My develop environment of Jetson TX1 is JetPack 2.3.1 / L4T R24.2.1 aarch64 (Ubuntu 16.04 LTS).

raaka1, 2017-02-23:
Are you doing cross platform compilation ?
> nvcc fatal : Cannot compile in the 32-bit mode when the host compiler targets aarch64.

Jia-HongHenryLee, 2017-02-23:
@raaka1 Thank your reply. I am not sure. Maybe not. I just let my my develop environment of Jetson TX1 become JetPack 2.3.1 / L4T R24.2.1 aarch64 (Ubuntu 16.04 LTS) , git clone new jetson-inference source code in Jetson TX1 and follow this README file.

Jia-HongHenryLee, 2017-02-24:
@raaka1  how to setup cross platform compilation?

Jia-HongHenryLee, 2017-02-24:
@raaka1  Thank you very much. i solve this problem. 
solve :
rm -r jetson-inference
sudo git clone  git clone http://github.org/dusty-nv/jetson-inference
cd jetson-inference
mkdir build
cd build
sudo cmake ..
sudo make -j8

closed 2017-02-24 by Jia-HongHenryLee

jamesthesken, 2020-02-12:
Trying some cross-compilation and I see the same error. I'll try to update if I find anything