dusty-nv/jetson-inference #43: Caffe Parser: could not parse binary model file

dengsitao, 2017-02-24:
Hi
I was using a TX1 dev board, I have flashed OS on it with Jetpack 2.3.1, 
then downloaded code and built it successfully on the dev board.
but when I go to aarch64/bin/ to run ./imagenet-console orange_0.jpg output_0.jpg
I got the error Caffe Parser could not parser binary model file

like this

$ ./imagenet-console ./orange_0.jpg output_0.jpg
imagenet-console
  args (3):  0 [./imagenet-console]  1 [./orange_0.jpg]  2 [output_0.jpg]  

[GIE]  attempting to open cache file bvlc_googlenet.caffemodel.tensorcache
[GIE]  cache file not found, profiling network model
[GIE]  platform has FP16 support.
[GIE]  loading googlenet.prototxt bvlc_googlenet.caffemodel
Caffe Parser: could not parse binary model file
Could not parse model file
[GIE]  failed to parse caffe network
failed to load bvlc_googlenet.caffemodel
failed to load bvlc_googlenet.caffemodel
imageNet -- failed to initialize.
imagenet-console:   failed to initialize imageNet

Will it be the problem of my downloading of the bvlc_googlenet.caffemodel file?

dusty-nv, 2017-02-24:
Hi there, I think the downloading of caffemodel became corrupted.  Can you delete the build directory, and re-run cmake from the beginning?  Or you could look in CMakePreBuild.sh script to manually find the URL of the model (it downloads with wget).

This incident also makes me think maybe I should add SHA checksum as a feature to validate the models.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

dengsitao, 2017-02-27:
I am re-run it now,
Would you please help to put a md5sum checksum here so I can compare it after downloading?

Thanks

dengsitao, 2017-02-27:
Hi
I have removed the whole build folder and then re-run the configure, now it is OK.
I also deleted the file under data/networks/bvlc*

It is caused by the download files are corrupted.
Thanks.