dusty-nv/jetson-inference #44: segnet overlay takes much time

lbin, 2017-02-25:
When I run segnet, I got layer network time is 32ms and overlay time is 351ms. How could I reduce the time.

dusty-nv, 2017-02-25:
Hi Li, which SegNet model are you running?  BTW soon I will be optimizing the segmentation overlay with CUDA.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

lbin, 2017-02-26:
Hi dusty
I used my own model, and my output just the crop layer.

Thanks for you reply!

closed 2017-03-14 by lbin