dusty-nv/jetson-inference #47: [GIE] Error parsing layer type Deconvolution index 379

Jia-HongHenryLee, 2017-03-02:
Can everyone help me how to solve this following error when i want to use deconvolution layer?

i see the following website, it shows that GIE supports deconvolution layer, but i try to add deconvolution layer to my prototxt and build the new caffe model whose format is not like goolgenet, alexnet and detectnet. I get this error.

[GIE]  loading pvaNetClassifier.prototxt pvanet_frcnn_iter_100000.caffemodel
Caffe Parser: groups are not supported for deconvolutions
error parsing layer type Deconvolution index 379
[GIE]  failed to parse caffe network

Thank you very much.

Jia-HongHenryLee, 2017-03-02:
Thank everyone. i find the developer document about GIE. This document is located in your deployment device (e.g. jetson TX1). The path is /usr/share/doc/gie. 
Solution: The version one of GIE don't support the groups parameter of deconvolution layer.

closed 2017-03-02 by Jia-HongHenryLee

lbin, 2017-03-13:
Hi @Jia-HongHenryLee , could you tell me what is the `groups parameter of deconvo layer`?

Jia-HongHenryLee, 2017-03-14:
Hi @lbin , you can visit the following website. it can solve your question.
http://stackoverflow.com/questions/40872914/caffe-what-does-the-group-param-mean

lbin, 2017-03-14:
Thanks for the reference, @Jia-HongHenryLee !
This `group` param confused me. My segmentation model has deconv layer, and  I find the default value of the group is `1`, and the model can be load by GIE. Unfortunately,  my result is totally wrong.
By the way, this model in caffe could get right results.

Jia-HongHenryLee, 2017-03-14:
@lbin  yes, because TensorRT(GIE) don't support the group parameter of deconvolution layer. so said. i think that the second version of Tensor RT maybe will support it.

Jia-HongHenryLee, 2017-03-14:
you can try to modify segnet which is in jetson-inference to let it have the same accuracy with your version

lbin, 2017-03-14:
@Jia-HongHenryLee Thanks for the kindly reply. Now I am doing my work based on the Caffe.

Jia-HongHenryLee, 2017-03-14:
@lbin  You're welcome.