dusty-nv/jetson-inference #51: explanation on "mCoverageThreshold" and "threshold" parameters

ShervinAr, 2017-03-05:
Dear all,

Could someone please provide some explanation on the "mCoverageThreshold" and "threshold" parameters used in detectNet.cpp. These parameters seem to have big effects on detection results.

dusty-nv, 2017-03-05:
They set the detection thresholds.  Modify them as you see fit per your dataset.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

ShervinAr, 2017-03-05:
@dusty-nv  Thanks for your quick response. Is there any more explanation available on these parameters and how they affect the detection...
In addition, the "mCoverageThreshold" parameter seems to be set to 0.5f in line 45 of detectNet.cpp and used in line 221. However, changing this parameter on line 45 does not seem to have any effects on the results. But when I replaced it in line 221 with some fixed value (say "0.7f") then the effects appear, which seems a bit strange to me...
why changing this parameter on line 45 does not affect detection whereas replacing it in line 221 with some fixed value (other than 0.5f) changes the results?

ShervinAr, 2017-03-05:
@dusty-nv  One more thing, how does the Digits environment set these parameters for a particular trained network?

closed 2023-04-27 by dusty-nv