dusty-nv/jetson-inference #53: Followed the instructions... cmake build overwhelms storage

MikeTam1021, 2017-03-08:
Hi, following the instructions of this repo, the build for cmake produces a file that spans into several gigabytes. I was not able to use the "make" command on this file afterwards since there was no makefile. I believe I ran out of space on the Jetson TX-1.

dusty-nv, 2017-03-08:
Hi, can you go into CMakePreBuild.sh and comment out some of the model files that you don’t think you’ll need?

In particular, the segmentation models that are downloaded near the end of the script are quite large and can be commented out if you aren’t running segNet.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

dusty-nv, 2017-03-08:
BTW, were you working off a fresh JetPack 2.3 install, or had you installed some things to your TX1 storage before building jetson-inference?   If the former, I will comment out some models for everyone.

MikeTam1021, 2017-03-08:
I was working off a fresh jetpack install when I did a cmake for this.

dusty-nv, 2017-03-08:
OK, thanks for letting me know, it should be fixed by commit [5cc77d](https://github.com/dusty-nv/jetson-inference/commit/5cc77d2f4a475521c69a0b34881e9f187d2840cd)

I would recommend deleting your jetson-inference directory and re-cloning.  If you are still having trouble with running out of storage, please let me know.

MikeTam1021, 2017-03-08:
will do

closed 2017-03-08 by MikeTam1021

MikeTam1021, 2017-03-08:
I ended up commenting out the other Alexnet Ariel and Cityscapes. It brought down from over 2 GB to 880 Mb