dusty-nv/jetson-inference #54: platform does not have FP16 support.

ShervinAr, 2017-03-09:
Hello,

I can compile and run the Jetson codes on my Desktop machine. However, during the run-time, I get the following message:
"_platform does not have FP16 support._"

**1-** I am using a Geforce GTX 950. If my platform does not support FP16, does this mean that it switches to FP32? and if yes, how does this affect my inference speed?
**2-** Does the Jetson TX1 module support FP16? 
**3-** does using  FP16 typically double inference speed compared to FP32?

many thanks

dusty-nv, 2017-03-09:
1. It should fall back to FP32, yes. That particular printf is a status message, not a terminal error. Or it shouldn't be anyway. Does the program continue to run? FP16 generally doubles performance.

2. Jetson TX1 and Jetson TX2 support FP16.

3. Yes.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

ShervinAr, 2017-03-09:
@dusty-nv  many thanks for your quick response. 

Yes,  the program continues to run.

4- Can one reduce the precision further to int8 on Jetson TX1/TX2 using TensorRT?

5- is there any advantage in looking into TensorFlow to achieve inference speed-up on Jetson TX1/TX2 or the tensorRT just does the same thing for us? I mean, is one able to achieve any inference speed-up over that of TensorRT using TensorFlow? (I am asking this as I guess TensorFlow might be used to generate int8 precision networks whereas TensorRT can't)

dusty-nv, 2017-03-09:
4. Not on Jetson, but is supported with TensorRT2 on dGPUs that support it: https://developer.nvidia.com/tensorrt

5. TensorFlow is a challenge to build on ARM and reports suggest performance is behind.  https://github.com/tensorflow/tensorflow/issues/851

http://stackoverflow.com/questions/39783919/tensorflow-on-nvidia-tx1

Personally I use TensorRT and Torch when I'm doing reinforcement learning. I hope pyTorch has low overhead like LUA Torch..

closed 2023-04-27 by dusty-nv