dusty-nv/jetson-inference #56: Deconvolution groups

lbin, 2017-03-13:
Hi @dusty-nv 
In gie's doc said, it doesnot supprot Deconvolution groups, pls give me some reference about what is this layer.
Thanks.

dusty-nv, 2017-03-13:
Hi Li,

Is this pertaining to FCN-Alexnet?  The deconv groups should be removed to get it to load with TensorRT.   Actually I remove the whole deconv layer, because it has learning rate = zero (i.e. is a simple linear upscaling) and is pretty slow.

Hope that helps,
Dustin

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

lbin, 2017-03-13:
Thanks for you reply, dusty!
My model has some different with FCN_Alexnet, so I must use deconv layer. I could load my model with deconv layer in TensorRT, but I got error results from this layer.  My deconv layer doesnot have `group` param, so what is `deconv groups`?

closed 2017-03-24 by lbin