dusty-nv/jetson-inference #60: Changing Camera source from onboard camera to video file

sevico, 2017-03-29:
Would it be possible to change the camera source  camera to video file?

mescarra, 2017-04-20:
This feature would be awesome

dawnhowe-sync, 2017-09-15:
I have made modifications to do this using a raw video file. Here's how I did it:

1. Video file should be in raw nv12 format.  I have some video from my video camera in a file named myvideo.MTS. Use ffmpeg to convert it (or you can use Jetpack 3.1 ~/tegra_multimedia_api/samples/00_video_decode)

`$ ffmpeg -i myvideo.MTS -vcodec rawvideo -pix_fmt nv12 myfilename.yuv`

It plays fine on a shell command line with:
`$ ffplay -f rawvideo -pix_fmt nv12 -s 1920x1080 myfilename.yuv`

2. Install the gstreamer plugins (I think videoparse is with the bad plugins, so this may be overkill):
`$ sudo apt-get install libgstreamer1.0-0 gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly gstreamer1.0-libav
gstreamer1.0-doc gstreamer1.0-tools`

3. Modify  gstCamera::buildLaunchStr()
`filesrc location=myfilename.yuv ! videoparse width=1920 height=1080 framerate=30/1 format=2 ! appsink name=mysink`

SaddamBInSyed, 2019-09-01:
Hi there.
Is there any update in latest Jetpack version. 

I am running code sample in Jetson nano. but 
I too couldn't able to load video file.

much needed, any help.

thanks