dusty-nv/jetson-inference #63: Compiling on host.

bpinaya, 2017-03-29:
Hi there Dustin, I am trying to compile this on a host computer just to see the TensorRT examples, been looking around and this repo seems to have it the clearest. 
I am trying on the master branch, with cuda 8 and tensorRT installed and I am seeing errors like these:
`<command-line>:0:7: warning: ISO C99 requires whitespace after the macro name [enabled by default]
<command-line>:0:7: warning: ISO C99 requires whitespace after the macro name [enabled by default]
<command-line>:0:7: warning: ISO C99 requires whitespace after the macro name [enabled by default]
<command-line>:0:7: warning: ISO C99 requires whitespace after the macro name [enabled by default]
/usr/lib/gcc/x86_64-linux-gnu/4.8/include/stddef.h(432): error: identifier "nullptr" is undefined

/usr/lib/gcc/x86_64-linux-gnu/4.8/include/stddef.h(432): error: expected a ";"

/usr/include/x86_64-linux-gnu/c++/4.8/bits/c++config.h(190): error: expected a ";"

/usr/include/c++/4.8/exception(63): error: expected a ";"

/usr/include/c++/4.8/exception(68): error: expected a ";"
`
I assume those are some C++11 errors but I saw the CmakeList and found:
`set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")`
Any tips I could use? All help is appreciated.

abdo-abaco, 2017-03-29:
Hi bpinaya, look for the CUDA_NVCC_FLAGS and add -std=c++11 as follows. This was shown on another comment.

set(
	CUDA_NVCC_FLAGS
	${CUDA_NVCC_FLAGS}; 
    -O3 
	-std=c++11 -gencode arch=compute_53,code=sm_53
	-gencode arch=compute_62,code=sm_62
)

bpinaya, 2017-03-29:
@abdo-abaco thanks for the quick response, I tried that one but I am still getting stuck at:
```
/usr/include/c++/4.8/type_traits(269): error: identifier "__float128" is undefined
1 error detected in the compilation of "/tmp/tmpxft_0000592f_00000000-9_cudaFont.compute_62.cpp1.ii".
CMake Error at jetson-inference_generated_cudaFont.cu.o.cmake:264 (message):
```
I have a tx1 where I was successfully running this but in the host computer it fails. If you know any other tensorRT examples I'd be thankful!

hitspring2015, 2017-03-31:
I am also trying to compile this on a host computer. What the IDE is Nsight  Eclipse Edition from Nvidia and the host computer is ubuntu14.04 .and I connect Tx1 and the host with ssh 

according to the follows what is from Cmakelist,I include the libraries and the .h 

``# build C/C++ interface
include_directories(${PROJECT_INCLUDE_DIR} ${GIE_PATH}/include)
include_directories(/usr/include/gstreamer-1.0 /usr/lib/aarch64-linux-gnu/gstreamer-1.0/include /usr/include/glib-2.0 /usr/include/libxml2 /usr/lib/aarch64-linux-gnu/glib-2.0/include/)

file(GLOB inferenceSources *.cpp *.cu util/*.cpp util/camera/*.cpp util/cuda/*.cu util/display/*.cpp)
file(GLOB inferenceIncludes *.h util/*.h util/camera/*.h util/cuda/*.h util/display/*.h)

cuda_add_library(jetson-inference SHARED ${inferenceSources})
target_link_libraries(jetson-inference nvcaffe_parser nvinfer Qt4::QtGui GL GLEW gstreamer-1.0 gstapp-1.0)		# gstreamer-0.10 gstbase-0.10 gstapp-0.10 

but now i can not compile it, what the error is 
Description	Resource	Path	Location	Type
./glDisplay.o: undefined reference to symbol 'XGetWindowAttributes'	test4face		 	C/C++ Problem

abdo-abaco, 2017-04-20:
@bpinaya it should compile with the correct libraries installed. Make sure TensorRT is installed and try https://github.com/Abaco-Systems/jetson-inference-gv. This has webcam support. This should compile as well and currently working on getting it to run on host.

abdo-abaco, 2017-04-20:
I got this demo to run on the host. So you don't need the Abaco one.
First step is to get it to compile. Then to get it to run:
 Run your deviceQuery example and find out your compute architecture. In my case 5.0 and 6.3. As mentioned by Dustin make to set #define DEFAULT_CAMERA to 0 on the detectnet-camera.cpp file.

Go to CMakeLists.txt and add the lines underneath: -gencode arch=compute_50,code=sm_50
-gencode arch=compute_50,code=sm_50
-gencode arch=compute_61,code=sm_61

Another note: make sure you have the host tools from jetpack installed. Not sure if this can help.

sulth, 2017-06-09:
admin@apnsfoml01:~/jetson-inference/build$ sudo make -j8
[  5%] Building CXX object CMakeFiles/jetson-inference.dir/imageNet.cpp.o
[  5%] Building CXX object CMakeFiles/jetson-inference.dir/detectNet.cpp.o
[  8%] Building CXX object CMakeFiles/jetson-inference.dir/segNet.cpp.o
[ 10%] Building CXX object CMakeFiles/jetson-inference.dir/util/camera/gstUtility.cpp.o
In file included from /home/admin/jetson-inference/detectNet.h:9:0,
                 from /home/admin/jetson-inference/detectNet.cpp:5:
/home/admin/jetson-inference/tensorNet.h:9:21: fatal error: NvInfer.h: No such file or directory
 #include "NvInfer.h"
                     ^
compilation terminated.
In file included from /home/admin/jetson-inference/segNet.h:9:0,
                 from /home/admin/jetson-inference/segNet.cpp:5:
/home/admin/jetson-inference/tensorNet.h:9:21: fatal error: NvInfer.h: No such file or directory
 #include "NvInfer.h"
                     ^
compilation terminated.
CMakeFiles/jetson-inference.dir/build.make:1373: recipe for target 'CMakeFiles/jetson-inference.dir/detectNet.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/detectNet.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /usr/include/glib-2.0/glib/galloca.h:32:0,
                 from /usr/include/glib-2.0/glib.h:30,
                 from /usr/include/gstreamer-1.0/gst/gst.h:27,
                 from /home/admin/jetson-inference/util/camera/gstUtility.h:9,
                 from /home/admin/jetson-inference/util/camera/gstUtility.cpp:5:
/usr/include/glib-2.0/glib/gtypes.h:32:24: fatal error: glibconfig.h: No such file or directory
[ 13%] Building CXX object CMakeFiles/jetson-inference.dir/tensorNet.cpp.o
compilation terminated.
CMakeFiles/jetson-inference.dir/build.make:1421: recipe for target 'CMakeFiles/jetson-inference.dir/segNet.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/segNet.cpp.o] Error 1
In file included from /home/admin/jetson-inference/imageNet.h:9:0,
                 from /home/admin/jetson-inference/imageNet.cpp:5:
/home/admin/jetson-inference/tensorNet.h:9:21: fatal error: NvInfer.h: No such file or directory
 #include "NvInfer.h"
                     ^
compilation terminated.
CMakeFiles/jetson-inference.dir/build.make:1517: recipe for target 'CMakeFiles/jetson-inference.dir/util/camera/gstUtility.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/util/camera/gstUtility.cpp.o] Error 1
CMakeFiles/jetson-inference.dir/build.make:1397: recipe for target 'CMakeFiles/jetson-inference.dir/imageNet.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/imageNet.cpp.o] Error 1
[ 16%] Building CXX object CMakeFiles/jetson-inference.dir/util/camera/gstCamera.cpp.o
In file included from /home/admin/jetson-inference/tensorNet.cpp:5:0:
/home/admin/jetson-inference/tensorNet.h:9:21: fatal error: NvInfer.h: No such file or directory
 #include "NvInfer.h"
                     ^
compilation terminated.
CMakeFiles/jetson-inference.dir/build.make:1445: recipe for target 'CMakeFiles/jetson-inference.dir/tensorNet.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/tensorNet.cpp.o] Error 1
In file included from /usr/include/glib-2.0/glib/galloca.h:32:0,
                 from /usr/include/glib-2.0/glib.h:30,
                 from /usr/include/gstreamer-1.0/gst/gst.h:27,
                 from /home/admin/jetson-inference/util/camera/gstCamera.h:8,
                 from /home/admin/jetson-inference/util/camera/gstCamera.cpp:5:
/usr/include/glib-2.0/glib/gtypes.h:32:24: fatal error: glibconfig.h: No such file or directory
compilation terminated.
CMakeFiles/jetson-inference.dir/build.make:1565: recipe for target 'CMakeFiles/jetson-inference.dir/util/camera/gstCamera.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/util/camera/gstCamera.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/jetson-inference.dir/all' failed
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

This is the error while iam compiling jetson inference.Please someone help me in finding a way ut.

dusty-nv, 2017-06-09:
Running jetson-inference on the host (which is not officially supported), you are missing NVIDIA TensorRT  (https://developer.nvidia.com/tensorrt) and gstreamer.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

sulth, 2017-06-09:
Thankyou.So whether i should install the jetson inference inside the board than in the host pc.

sulth, 2017-06-09:
Now i tried installing jetson inference on jetson tx2 board,i have errors as mentioned down.Please may help me build it in cmake ../
FCN-Alexnet-Aerial-FPV-720p/snapshot_iter_10280.caffemodel
FCN-Alexnet-Aerial-FPV-720p/fpv-labels.txt
FCN-Alexnet-Aerial-FPV-720p/fpv-deploy-colors.txt
FCN-Alexnet-Aerial-FPV-720p/fcn_alexnet.deploy.prototxt
FCN-Alexnet-Aerial-FPV-720p/fcn_alexnet.digits.deploy.prototxt
FCN-Alexnet-Aerial-FPV-720p/fpv-training-colors.txt
[Pre-build]  Finished CMakePreBuild script
Finished installing dependencies
qmake: could not exec '/usr/lib/aarch64-linux-gnu/qt4/bin/qmake': No such file or directory
CMake Error at /usr/share/cmake-3.5/Modules/FindQt4.cmake:1326 (message):
  Found unsuitable Qt version "" from NOTFOUND, this code requires Qt 4.x
Call Stack (most recent call first):
  CMakeLists.txt:24 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/nvidia/jetson-inference/build/CMakeFiles/CMakeOutput.log".
nvidia@tegra-ubuntu:~/jetson-inference/build$

dusty-nv, 2017-06-09:
OK, it looks like qt dev package is missing.  Can you try running this from the command line:

`sudo apt-get install -y libqt4-dev qt4-dev-tools libglew-dev glew-utils libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev libglib2.0-dev`

Also this should have already been run in `CMakePreBuild.sh` script during `cmake ../` step… you may want to delete build directory and try again.

sulth, 2017-06-09:
Thanks i have installed all those packages except  Package 'glew-utils' has no installation candidate in the location/usr/lib/aarch64-linux-gnu
  ,now iam getting  error  in cmake ../ as this.please look into it..............................................................................................................................................................FCN-Alexnet-Aerial-FPV-720p/fpv-training-colors.txt
[Pre-build]  Finished CMakePreBuild script
Finished installing dependencies
-- Looking for Q_WS_X11
-- Looking for Q_WS_X11 - found
-- Looking for Q_WS_WIN
-- Looking for Q_WS_WIN - not found
-- Looking for Q_WS_QWS
-- Looking for Q_WS_QWS - not found
-- Looking for Q_WS_MAC
-- Looking for Q_WS_MAC - not found
-- Found Qt4: /usr/bin/qmake (found version "4.8.7")
CUDA_TOOLKIT_ROOT_DIR not found or specified
-- Could NOT find CUDA (missing:  CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY)
-- system arch:  aarch64...........................................................................................................ra/v4l2-display
   used as include directory in directory /home/nvidia/jetson-inference/util/camera/v4l2-display
   used as include directory in directory /home/nvidia/jetson-inference/docs
   used as include directory in directory /home/nvidia/jetson-inference/docs
   used as include directory in directory /home/nvidia/jetson-inference/docs
   used as include directory in directory /home/nvidia/jetson-inference/docs
   used as include directory in directory /home/nvidia/jetson-inference/docs
   used as include directory in directory /home/nvidia/jetson-inference/docs

-- Configuring incomplete, errors occurred!

sulth, 2017-06-10:
CMake Error: The following variables are used in this project, but they are set                                                                                         to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake file                                                                                        s:
CUDA_CUDART_LIBRARY (ADVANCED)
    linked by target "jetson-inference" in directory /home/nvidia/jetson-inferen                                                                                        ce
    linked by target "imagenet-console" in directory /home/nvidia/jetson-inferen                                                                                        ce/imagenet-console
    linked by target "imagenet-camera" in directory /home/nvidia/jetson-inferenc                                                                                        e/imagenet-camera
    linked by target "detectnet-console" in directory /home/nvidia/jetson-infere                                                                                        nce/detectnet-console
    linked by target "detectnet-camera" in directory /home/nvidia/jetson-inferen                                                                                        ce/detectnet-camera
    linked by target "segnet-console" in directory /home/nvidia/jetson-inference                                                                                        /segnet-console
CUDA_TOOLKIT_INCLUDE (ADVANCED)
   used as include directory in directory /home/nvidia/jetson-inference
   used as include directory in directory /home/nvidia/jetson-inference
   used as include directory in directory /home/nvidia/jetson-inference
   used as include directory in directory /home/nvidia/jetson-inference
   used as include directory in directory /home/nvidia/jetson-inference
   used as include directory in directory /home/nvidia/jetson-inference
   used as include directory in directory /home/nvidia/jetson-inference
   used as include directory in directory /home/nvidia/jetson-inference/imagenet                                                                                        -console
   used as include directory in directory /home/nvidia/jetson-inference/imagenet                                                                                        -console
   used as include directory in directory /home/nvidia/jetson-inference/imagenet                                                                                        -console
   used as include directory in directory /home/nvidia/jetson-inference/imagenet                                                                                        -console
   used as include directory in directory /home/nvidia/jetson-inference/imagenet                                                                                        -console
   used as include directory in directory /home/nvidia/jetson-inference/imagenet                                                                                        -console
   used as include directory in directory /home/nvidia/jetson-inference/imagenet                                                                                        -console
   used as include directory in directory /home/nvidia/jetson-inference/imagenet                                                                                        -console
   used as include directory in directory /home/nvidia/jetson-inference/imagenet                                                                                        -console
   used as include directory in directory /home/nvidia/jetson-inference/imagenet                                                                                        -console
   used as include directory in directory /home/nvidia/jetson-inference/imagenet                                                                                        -console
   used as include directory in directory /home/nvidia/jetson-inference/imagenet                                                                                        -console
   used as include directory in directory /home/nvidia/jetson-inference/imagenet                                                                                        -console
   used as include directory in directory /home/nvidia/jetson-inference/imagenet                                                                                        -console
   used as include directory in directory /home/nvidia/jetson-inference/imagenet                                                                                        -camera
   used as include directory in directory /home/nvidia/jetson-inference/imagenet                                                                                        -camera
   used as include directory in directory /home/nvidia/jetson-inference/imagenet                                                                                        -camera
   used as include directory in directory /home/nvidia/jetson-inference/imagenet                                                                                        -camera
   used as include directory in directory /home/nvidia/jetson-inference/imagenet                                                                                        -camera
[... cut ...]
   used as include directory in directory /home/nvidia/jetson-inference/docs
   used as include directory in directory /home/nvidia/jetson-inference/docs
   used as include directory in directory /home/nvidia/jetson-inference/docs

-- Configuring incomplete, errors occurred!
See also "/home/nvidia/jetson-inference/build/CMakeFiles/CMakeOutput.log".
See also "/home/nvidia/jetson-inference/build/CMakeFiles/CMakeError.log".

Please may help me analyse the problem.Iam stuck.Is it with any problem on jetpack installation.

limcheling, 2018-03-14:
Hi, any follow up? I have the same problem too and stuck here.

trohit920, 2018-07-27:
Hello, I am trying to run Jetson-Inference on a host machine and got stuck with this error. I have installed TensorRT and GStreamer and all other Cuda related requirements. Please can you see what i am doing wrong here.

`~/jetson-inference/build$ sudo make -j8
-- system arch:  x86_64
-- output path:  /home/rohit/jetson-inference/build/x86_64
-- Copying /home/rohit/jetson-inference/tensorNet.h
-- Copying /home/rohit/jetson-inference/imageNet.h
-- Copying /home/rohit/jetson-inference/detectNet.h
-- Copying /home/rohit/jetson-inference/segNet.h
-- Copying /home/rohit/jetson-inference/util/loadImage.h
-- Copying /home/rohit/jetson-inference/util/commandLine.h
-- Copying /home/rohit/jetson-inference/util/camera/v4l2Camera.h
-- Copying /home/rohit/jetson-inference/util/camera/gstCamera.h
-- Copying /home/rohit/jetson-inference/util/camera/gstUtility.h
-- Copying /home/rohit/jetson-inference/util/cuda/cudaRGB.h
-- Copying /home/rohit/jetson-inference/util/cuda/cudaOverlay.h
-- Copying /home/rohit/jetson-inference/util/cuda/cudaUtility.h
-- Copying /home/rohit/jetson-inference/util/cuda/cudaYUV.h
-- Copying /home/rohit/jetson-inference/util/cuda/cudaResize.h
-- Copying /home/rohit/jetson-inference/util/cuda/cudaFont.h
-- Copying /home/rohit/jetson-inference/util/cuda/cudaNormalize.h
-- Copying /home/rohit/jetson-inference/util/cuda/cudaMappedMemory.h
-- Copying /home/rohit/jetson-inference/util/display/glUtility.h
-- Copying /home/rohit/jetson-inference/util/display/glTexture.h
-- Copying /home/rohit/jetson-inference/util/display/glDisplay.h
-- Copying /home/rohit/jetson-inference/data/images/drone_0436.png
-- Copying /home/rohit/jetson-inference/data/images/dog_0.jpg
-- Copying /home/rohit/jetson-inference/data/images/dog_2.jpg
-- Copying /home/rohit/jetson-inference/data/images/drone_0255.png
-- Copying /home/rohit/jetson-inference/data/images/red_apple_0.jpg
-- Copying /home/rohit/jetson-inference/data/images/drone_0427.png
-- Copying /home/rohit/jetson-inference/data/images/bird_0.jpg
-- Copying /home/rohit/jetson-inference/data/images/granny_smith_1.jpg
-- Copying /home/rohit/jetson-inference/data/images/bottle_0.jpg
-- Copying /home/rohit/jetson-inference/data/images/peds-002.jpg
-- Copying /home/rohit/jetson-inference/data/images/dog_1.jpg
-- Copying /home/rohit/jetson-inference/data/images/fontmapA.png
-- Copying /home/rohit/jetson-inference/data/images/peds-004.jpg
-- Copying /home/rohit/jetson-inference/data/images/granny_smith_0.jpg
-- Copying /home/rohit/jetson-inference/data/images/peds-001.jpg
-- Copying /home/rohit/jetson-inference/data/images/airplane_0.jpg
-- Copying /home/rohit/jetson-inference/data/images/cat_0.jpg
-- Copying /home/rohit/jetson-inference/data/images/peds-003.jpg
-- Copying /home/rohit/jetson-inference/data/images/fontmapB.png
-- Copying /home/rohit/jetson-inference/data/images/orange_0.jpg
-- Copying /home/rohit/jetson-inference/data/images/banana_0.jpg
-- Copying /home/rohit/jetson-inference/data/images/orange_1.jpg
-- Copying /home/rohit/jetson-inference/data/images/drone_0435.png
-- Copying /home/rohit/jetson-inference/data/images/drone_0428.png
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/rohit/jetson-inference/build
[  2%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/util/cuda/jetson-inference_generated_cudaOverlay.cu.o
[  5%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/util/cuda/jetson-inference_generated_cudaResize.cu.o
[  7%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/util/cuda/jetson-inference_generated_cudaRGB.cu.o
[ 10%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/util/cuda/jetson-inference_generated_cudaYUV-NV12.cu.o
[ 12%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/util/cuda/jetson-inference_generated_cudaYUV-YV12.cu.o
[ 15%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/util/cuda/jetson-inference_generated_cudaFont.cu.o
[ 17%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/jetson-inference_generated_imageNet.cu.o
[ 20%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/util/cuda/jetson-inference_generated_cudaNormalize.cu.o
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
[... 447 lines cut ...]
/home/rohit/jetson-inference/tensorNet.cpp: In member function ‘bool tensorNet::LoadNetwork(const char*, const char*, const char*, const char*, int)’:
/home/rohit/jetson-inference/tensorNet.cpp:220:115: error: ‘NV_GIE_VERSION’ was not declared in this scope
 ild %u\n", NV_TENSORRT_MAJOR, NV_TENSORRT_MINOR, NV_TENSORRT_PATCH, NV_GIE_VERS
                                                                     ^
[... cut ...]
CMakeFiles/jetson-inference.dir/build.make:1457: recipe for target 'CMakeFiles/jetson-inference.dir/imageNet.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/imageNet.cpp.o] Error 1
CMakeFiles/jetson-inference.dir/build.make:1409: recipe for target 'CMakeFiles/jetson-inference.dir/tensorNet.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/tensorNet.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/jetson-inference.dir/all' failed
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
`

bpinaya, 2018-07-27:
@trohit920 it should work with @abdo-abaco tips, at the end I got it compiling on a 16.04 with a fresh install. Also I think that to better get a grasp on TensorRT it's a good idea to check their [dev-guide](https://docs.nvidia.com/deeplearning/sdk/pdf/TensorRT-Developer-Guide.pdf)

closed 2018-07-27 by bpinaya

trohit920, 2018-07-30:
Thanks for your reply but its not working for me..!

bpinaya, 2018-07-30:
Did you use c++11 to compile your code? As per the suggestions above? Some of the errors you have to fix yourself, just check the output, for example:
```
In file included from /home/rohit/jetson-inference/tensorNet.cpp:23:0:
/home/rohit/jetson-inference/tensorNet.h:209:7: error: ‘vector’ in namespace ‘std’ does not name a template type
  std::vector<outputLayer> mOutputs;
```
For that check [this link](https://stackoverflow.com/questions/16868230/vector-in-namespace-std-does-not-name-a-type). You just have to include the vector library as:
```c++
#include <vector>
```

Or your first error:

```
[ 46%] Building CXX object CMakeFiles/jetson-inference.dir/util/camera/v4l2Camera.cpp.o
/home/rohit/jetson-inference/util/camera/gstUtility.cpp: In function ‘gboolean gst_message_print(GstBus*, GstMessage*, gpointer)’:
/home/rohit/jetson-inference/util/camera/gstUtility.cpp:221:17: warning: ISO C++ forbids converting a string constant to ‘gchar* {aka char*}’ [-Wwrite-strings]
    gchar* txt = "missing gst_tag_list_to_string()";
```

That one is regarding a conversion from a string to a gchar, you can find [more here](https://www.linuxquestions.org/questions/programming-9/string-to-gchar%2A-487178/) but basically, you'd have to do use `.c_str()` to properly pass it to gchar.
Do check out @abdo-abaco repo [here](https://github.com/Abaco-Systems/jetson-inference-gv) I've seen it through and it seems that has some fixes to get it to compile on host.

I remember I was struggling with getting it to compile on host but it depends on what you want to do with it, I just wanted to learn more about tensorRt and ended up using their dev-guide and build my own examples. If you want to use object detection with tensorrt then it might be worth while to get it to compile on host, but if you just want to learn more about tensorRT then I recommend you just use the code here, the dev-guide and go from there.

Good luck!

bpinaya, 2018-08-06:
@trohit920 I just cloned the repo again and built it on host, it seems its easier to build now, I'll post some notes so that you can build it yourself.
Before, this is my `nvcc --version` (shouldn't matter that much):
```
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2018 NVIDIA Corporation
Built on Wed_Apr_11_23:16:29_CDT_2018
Cuda compilation tools, release 9.2, V9.2.88
```

My `cmake --version`:
```
cmake version 3.5.1
```
And my `gcc --version`:

```
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE
```

To build on host I had to make the following extra installs:
- Gstreamer: To get video input the repo uses gstreamer, check out the install steps [here](https://gstreamer.freedesktop.org/documentation/installing/on-linux.html) or just run
`sudo apt-get install libgstreamer1.0-0 gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly gstreamer1.0-libav gstreamer1.0-doc gstreamer1.0-tools`
- Glib2-0: Install from [launchpad](https://launchpad.net/ubuntu/+source/glib2.0)

Afterward you should cmake and make and you'll have some issues with those two libraries not being found. if you check this section of the `CMakeLists.txt`  it says:
```Cmake
# build C/C++ interface
include_directories(${PROJECT_INCLUDE_DIR} ${GIE_PATH}/include)
include_directories(/usr/include/gstreamer-1.0 /usr/lib/aarch64-linux-gnu/gstreamer-1.0/include /usr/include/glib-2.0 /usr/include/libxml2 /usr/lib/aarch64-linux-gnu/glib-2.0/include/)

file(GLOB inferenceSources *.cpp *.cu util/*.cpp util/camera/*.cpp util/cuda/*.cu util/display/*.cpp)
file(GLOB inferenceIncludes *.h util/*.h util/camera/*.h util/cuda/*.h util/display/*.h)

cuda_add_library(jetson-inference SHARED ${inferenceSources})
target_link_libraries(jetson-inference nvcaffe_parser nvinfer Qt4::QtGui GL GLEW gstreamer-1.0 gstapp-1.0)		# gstreamer-0.10 gstbase-0.10 gstapp-0.10 

```
So you see it's looking for `glib` and `gstreamer` in the `/usr/lib/aarch64-linux-gnu/`folder, since you are on host that probably means you are on x86_64, so change that `aarch64-linux-gnu` for `x86_64-linux-gnu` on both of those places.
After that it build normally, I'm attaching a screenshot to prove it hehe

[image]

If you have further problems building it let me know, hopefully this helps other folks trying to build on host. Which is useful if you don't have a jetson on hand and want to try TensorRT out.

bpinaya, 2018-08-06:
And I have a 1080Ti, and CUDA 9.2 but it shouldn't matter much.

bpinaya, 2018-08-06:
Just to add to this, I changed my arch and compute to 61 instead of 61 (Since I have the 1080Ti), rebuilt and works perfectly.
Two images showing inference on host, the first one on the `peds-001.jpg` image:
[image]

And this second one on an image from the Cityscapes dataset (since they use it for training):
[image]

Yao1up, 2020-09-26:
I  successfully ran the command of $cmake  ../  in  /build file folder, and the result of this step shows "Build files have been written to: /home/dc2-user/jetson-inference/build", but when I run $make, it told me that the function 'gst_structure_get_list'  was not declared.
compile error:
[ 12%] Building CXX object utils/CMakeFiles/jetson-utils.dir/URI.cpp.o
[ 13%] Building CXX object utils/CMakeFiles/jetson-utils.dir/XML.cpp.o
[ 14%] Building CXX object utils/CMakeFiles/jetson-utils.dir/camera/gstCamera.cpp.o
/home/dc2-user/jetson-inference/utils/camera/gstCamera.cpp: In member function ‘bool gstCamera::parseCaps(GstStructure*, videoOptions::Codec*, imageFormat*, uint32_t*, uint32_t*, float*)’:
/home/dc2-user/jetson-inference/utils/camera/gstCamera.cpp:259:63: error: ‘gst_structure_get_list’ was not declared in this scope
   if( gst_structure_get_list(caps, "framerate", &frameRateList) && frameRateList->n_values > 0 )
                                                               ^
utils/CMakeFiles/jetson-utils.dir/build.make:200: recipe for target 'utils/CMakeFiles/jetson-utils.dir/camera/gstCamera.cpp.o' failed
make[2]: *** [utils/CMakeFiles/jetson-utils.dir/camera/gstCamera.cpp.o] Error 1
CMakeFiles/Makefile2:750: recipe for target 'utils/CMakeFiles/jetson-utils.dir/all' failed
make[1]: *** [utils/CMakeFiles/jetson-utils.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2