dusty-nv/jetson-inference #64: USB Camera on TX2

S4WRXTTCS, 2017-04-04:
Has anyone tested using detectnet-camera on the TX2 with a USB camera?  

I've tried a couple cameras that work fine on with detectnet-camera on the TX1, but they don't work with detectnet-camera on the TX2. They do work correctly with the cheese app.

Here is the error message it gives. 

nvidia@tegra-ubuntu:/media/nvidia/JetsonSSD/jetson-inference/build/aarch64/bin$ ./detectnet-camera
detectnet-camera
  args (1):  0 [./detectnet-camera]  

[gstreamer] initialized gstreamer, version 1.8.3.0
[gstreamer] gstreamer decoder pipeline string:
v4l2src device=/dev/video1 ! video/x-raw, width=(int)1280, height=(int)720, format=RGB ! videoconvert ! video/x-raw, format=RGB ! videoconvert !appsink name=mysink

detectnet-camera:  successfully initialized video device
    width:  1280
   height:  720
    depth:  24 (bpp)

[GIE]  attempting to open cache file multiped-500/snapshot_iter_178000.caffemodel.tensorcache
[GIE]  loading network profile from cache... multiped-500/snapshot_iter_178000.caffemodel.tensorcache
[GIE]  platform has FP16 support.
[GIE]  multiped-500/snapshot_iter_178000.caffemodel loaded
[GIE]  CUDA engine context initialized with 3 bindings
[GIE]  multiped-500/snapshot_iter_178000.caffemodel input  binding index:  0
[GIE]  multiped-500/snapshot_iter_178000.caffemodel input  dims (b=2 c=3 h=512 w=1024) size=12582912
[cuda]  cudaAllocMapped 12582912 bytes, CPU 0x102800000 GPU 0x102800000
[GIE]  multiped-500/snapshot_iter_178000.caffemodel output 0 coverage  binding index:  1
[GIE]  multiped-500/snapshot_iter_178000.caffemodel output 0 coverage  dims (b=2 c=2 h=32 w=64) size=32768
[cuda]  cudaAllocMapped 32768 bytes, CPU 0x103400000 GPU 0x103400000
[GIE]  multiped-500/snapshot_iter_178000.caffemodel output 1 bboxes  binding index:  2
[GIE]  multiped-500/snapshot_iter_178000.caffemodel output 1 bboxes  dims (b=2 c=4 h=32 w=64) size=65536
[cuda]  cudaAllocMapped 65536 bytes, CPU 0x103408000 GPU 0x103408000
multiped-500/snapshot_iter_178000.caffemodel initialized.
[cuda]  cudaAllocMapped 32 bytes, CPU 0x103600000 GPU 0x103600000
maximum bounding boxes:  8192
[cuda]  cudaAllocMapped 131072 bytes, CPU 0x103800000 GPU 0x103800000
[cuda]  cudaAllocMapped 65536 bytes, CPU 0x103418000 GPU 0x103418000
default X screen 0:   1920 x 1200
[OpenGL]  glDisplay display window initialized
[OpenGL]   creating 1280x720 texture
loaded image  fontmapA.png  (256 x 512)  2097152 bytes
[cuda]  cudaAllocMapped 2097152 bytes, CPU 0x103a00000 GPU 0x103a00000
[cuda]  cudaAllocMapped 8192 bytes, CPU 0x103c00000 GPU 0x103c00000
[gstreamer] gstreamer transitioning pipeline to GST_STATE_PLAYING
[gstreamer] gstreamer changed state from NULL to READY ==> mysink
[gstreamer] gstreamer changed state from NULL to READY ==> videoconvert1
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter1
[gstreamer] gstreamer changed state from NULL to READY ==> videoconvert0
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter0
[gstreamer] gstreamer changed state from NULL to READY ==> v4l2src0
[gstreamer] gstreamer changed state from NULL to READY ==> pipeline0
[gstreamer] gstreamer changed state from READY to PAUSED ==> videoconvert1
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter1
[gstreamer] gstreamer changed state from READY to PAUSED ==> videoconvert0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter0
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> v4l2src0
[gstreamer] gstreamer changed state from READY to PAUSED ==> pipeline0
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer msg new-clock ==> pipeline0
[gstreamer] gstreamer msg stream-start ==> pipeline0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> videoconvert1
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter1
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> videoconvert0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> v4l2src0
[gstreamer] gstreamer decoder onEOS
[gstreamer] gstreamer v4l2src0 ERROR Internal data flow error.
[gstreamer] gstreamer Debugging info: gstbasesrc.c(2948): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming task paused, reason not-negotiated (-4)
[gstreamer] gstreamer changed state from READY to PAUSED ==> mysink

detectnet-camera:  camera open for streaming

detectnet-camera:  failed to capture frame
detectnet-camera:  failed to convert from NV12 to RGBA
detectNet::Detect( 0x(nil), 1280, 720 ) -> invalid parameters
[cuda]   cudaNormalizeRGBA((float4*)imgRGBA, make_float2(0.0f, 255.0f), (float4*)imgRGBA, make_float2(0.0f, 1.0f), camera->GetWidth(), camera->GetHeight())
[cuda]      invalid device pointer (error 17) (hex 0x11)
[cuda]      /media/nvidia/JetsonSSD/jetson-inference/detectnet-camera/detectnet-camera.cpp:229
[cuda]   registered 14745600 byte openGL texture for interop access (1280x720)

detectnet-camera:  failed to capture frame
detectnet-camera:  failed to convert from NV12 to RGBA
detectNet::Detect( 0x(nil), 1280, 720 ) -> invalid parameters
[cuda]   cudaNormalizeRGBA((float4*)imgRGBA, make_float2(0.0f, 255.0f), (float4*)imgRGBA, make_float2(0.0f, 1.0f), camera->GetWidth(), camera->GetHeight())
[cuda]      invalid device pointer (error 17) (hex 0x11)
[cuda]      /media/nvidia/JetsonSSD/jetson-inference/detectnet-camera/detectnet-camera.cpp:229

S4WRXTTCS, 2017-04-04:
It works correctly using a USB 2.0 UVC camera, and it works correctly with a USB 3.0 UVC camera plugged into a different SuperSpeed port.

closed 2017-04-04 by S4WRXTTCS

fangtaosong, 2021-07-30:
how you solve this problem？