dusty-nv/jetson-inference #68: How to run compiling the imageNet-camera in the host（ubuntu14.04）and debug the executable on the TX1

hitspring2015, 2017-04-08:
hello, Now I have compiled the  imageNet-camera successfully on the nsight ,then I debug the remote application on the TX1 .But I cannot create a OpenGL window for display. The error is:

 ```
gender_google.caffemodel initialized.
[GIE]  gender loaded
imageNet -- loaded 2 class info entries
gender initialized.
[OpenGL]  failed to open X11 server connection.[OpenGL]  failed to create X11 Window.

imagenet-camera:  failed to create openGL display
failed to load image fontmapA.png
[gstreamer] gstreamer transitioning pipeline to GST_STATE_PLAYING

```

Also the display window don't be created on the Tx1.
hope your help ! thanks

nPellejero, 2017-04-19:
Hello,

I'm facing a similar problem, I need to run imagenet-camera from a ssh session, with X Forwarding. I get the same error. I tried installing virtualgl but nothing. Any thoughts? 

Thanks.

lakshpvt, 2018-12-19:
@nPellejero 
did you find any solution, even i have done the exact same thing and i do get the same error ..

is it because we have not connected a display using hdmi ??

closed 2023-04-27 by dusty-nv