dusty-nv/jetson-inference #72: cannot download model files in "nvidia.app.box.com"

gupengju, 2017-04-24:
## After I type "cmake .." in build dictory, it report  "Unable to locally verify the issuer's authority."
here is the log information:
...
Get:1 file:/var/cuda-repo-8-0-local  InRelease
Ign:1 file:/var/cuda-repo-8-0-local  InRelease
Get:2 file:/var/nv-gie-repo-ga-cuda8.0-gie1.0-20170116  InRelease
Ign:2 file:/var/nv-gie-repo-ga-cuda8.0-gie1.0-20170116  InRelease
Get:3 file:/var/cuda-repo-8-0-local  Release [574 B]
Get:4 file:/var/nv-gie-repo-ga-cuda8.0-gie1.0-20170116  Release [574 B]
Get:3 file:/var/cuda-repo-8-0-local  Release [574 B]
Get:4 file:/var/nv-gie-repo-ga-cuda8.0-gie1.0-20170116  Release [574 B]
Hit:7 http://ports.ubuntu.com/ubuntu-ports xenial InRelease
Hit:8 http://ports.ubuntu.com/ubuntu-ports xenial-updates InRelease
Hit:9 http://ports.ubuntu.com/ubuntu-ports xenial-security InRelease
Reading package lists...
W: Invalid 'Date' entry in Release file /var/lib/apt/lists/_var_nv-gie-repo-ga-cuda8.0-gie1.0-20170116_Release
--2017-04-24 23:29:40--  https://nvidia.box.com/shared/static/5j264j7mky11q8emy4q14w3r8hl5v6zh.caffemodel
Resolving nvidia.box.com (nvidia.box.com)... 107.152.24.197
Connecting to nvidia.box.com (nvidia.box.com)|107.152.24.197|:443... connected.
WARNING: cannot verify nvidia.box.com's certificate, issued by ‘CN=GeoTrust SSL CA - G4,O=GeoTrust Inc.,C=US’:
  Unable to locally verify the issuer's authority.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://nvidia.app.box.com/shared/static/5j264j7mky11q8emy4q14w3r8hl5v6zh.caffemodel [following]
--2017-04-24 23:29:41--  https://nvidia.app.box.com/shared/static/5j264j7mky11q8emy4q14w3r8hl5v6zh.caffemodel
Resolving nvidia.app.box.com (nvidia.app.box.com)... 8.7.198.45, 200:2:4e10:310f::
Connecting to nvidia.app.box.com (nvidia.app.box.com)|8.7.198.45|:443... failed: Connection timed out.
Connecting to nvidia.app.box.com (nvidia.app.box.com)|200:2:4e10:310f::|:443... failed: Network is unreachable.
--2017-04-24 23:31:48--  https://nvidia.box.com/shared/static/at8b1105ww1c5h7p30j5ko8qfnxrs0eg.caffemodel
Resolving nvidia.box.com (nvidia.box.com)... 107.152.25.197
Connecting to nvidia.box.com (nvidia.box.com)|107.152.25.197|:443... connected.
WARNING: cannot verify nvidia.box.com's certificate, issued by ‘CN=GeoTrust SSL CA - G4,O=GeoTrust Inc.,C=US’:
  Unable to locally verify the issuer's authority.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://nvidia.app.box.com/shared/static/at8b1105ww1c5h7p30j5ko8qfnxrs0eg.caffemodel [following]
--2017-04-24 23:31:49--  https://nvidia.app.box.com/shared/static/at8b1105ww1c5h7p30j5ko8qfnxrs0eg.caffemodel
Resolving nvidia.app.box.com (nvidia.app.box.com)... 8.7.198.45, 200:2:4e10:310f::
Connecting to nvidia.app.box.com (nvidia.app.box.com)|8.7.198.45|:443... 

## then I could compile it completely, but when run 
nvidia@tegra-ubuntu:~/Documents/mygithub/jetson-inference/build/aarch64/bin$ ./imagenet-console orange_0.jpg output_0.jpg
imagenet-console
  args (3):  0 [./imagenet-console]  1 [orange_0.jpg]  2 [output_0.jpg]  

[GIE]  attempting to open cache file networks/bvlc_googlenet.caffemodel.2.tensorcache
[GIE]  cache file not found, profiling network model
[GIE]  platform has FP16 support.
[GIE]  loading networks/googlenet.prototxt networks/bvlc_googlenet.caffemodel
[libprotobuf FATAL ../../../externals/protobuf/aarch64/10.0/include/google/protobuf/repeated_field.h:1408] CHECK failed: (index) < (current_size_): 
terminate called after throwing an instance of 'google_private::protobuf::FatalException'
  what():  CHECK failed: (index) < (current_size_): 
Aborted (core dumped)

dusty-nv, 2017-04-24:
Can you go into CMakePreBuild.sh, line 35-36, and try manually running the commands?  It is where the models are downloaded from box.com.  If it is unable to succeed you may have network connectivity problem or IT blockage.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

omaralvarez, 2017-05-25:
Same problem here. Either running the commands manually or from CMake does not work. The speed is really slow, also tried from other computer and from a browser, no go.

dusty-nv, 2017-05-25:
Hi, do the Google drive links in the script work for you?

omaralvarez, 2017-05-25:
Yes, only the Box links are failing.

omaralvarez, 2017-05-25:
I changed the firtst download links to Berkeley links and they work fine. Now stuck after the drive links...

dusty-nv, 2017-05-25:
OK, let me upload the files to Google drive for this thread.

dusty-nv, 2017-05-25:
Hi, here is Google drive link to model folder:  [https://drive.google.com/drive/folders/0BwYxpotGWRNOV2d1NXNJOHA0dzQ?usp=sharing](https://drive.google.com/drive/folders/0BwYxpotGWRNOV2d1NXNJOHA0dzQ?usp=sharing)

You can manually download and extract them to `jetson-inference/data/networks`
I can't use Google drive for the script because it doesn't permit automated dowloading (i.e. from `wget`)

omaralvarez, 2017-05-26:
Thanks dusty!

hfutzxy, 2017-06-16:
d

pooluut, 2017-06-29:
@dusty-nv 
could you please upload coco.tar.gz of the link "https://nvidia.box.com/shared/static/tdrvaw3fd2cwst2zu2jsi0u43vzk8ecu.gz"
to the google drive?

wojiaoxiaobei, 2017-08-02:
I have the same problem,when I follow the step to do, it doesn't work. how to do ?

dusty-nv, 2017-08-02:
Are you able to resolve the link to Google Drive above?  The models are backed up there.

wojiaoxiaobei, 2017-08-03:
yes,I have finished downloading。

then I could compile it completely, but when run

the result is 

nvidia@tegra-ubuntu:~/Documents/mygithub/jetson-inference/build/aarch64/bin$ ./imagenet-console orange_0.jpg output_0.jpg
imagenet-console
args (3): 0 [./imagenet-console] 1 [orange_0.jpg] 2 [output_0.jpg]

[GIE] attempting to open cache file networks/bvlc_googlenet.caffemodel.2.tensorcache
[GIE] cache file not found, profiling network model
[GIE] platform has FP16 support.
[GIE] loading networks/googlenet.prototxt networks/bvlc_googlenet.caffemodel
[libprotobuf FATAL ../../../externals/protobuf/aarch64/10.0/include/google/protobuf/repeated_field.h:1408] CHECK failed: (index) < (current_size_):
terminate called after throwing an instance of 'google_private::protobuf::FatalException'
what(): CHECK failed: (index) < (current_size_):
Aborted (core dumped)

wojiaoxiaobei, 2017-08-03:
I have solved the problem

gengqx, 2017-08-04:
Please pay attention to that the file in Google Drive does not include ‘bvlc_alexnet.caffemodel’ and 'bvle_googlenet.caffemodel'.Those two files are alse download from nvidia.app.box.com
```
but we can unconment CMakePreBuild.sh line 29 and 34
wget http://dl.caffe.berkeleyvision.org/bvlc_alexnet.caffemodel
wget http://dl.caffe.berkeleyvision.org/bvlc_googlenet.caffemodel
```
get those two files.

shellyfung, 2019-04-08:
> Hi, here is Google drive link to model folder: https://drive.google.com/drive/folders/0BwYxpotGWRNOV2d1NXNJOHA0dzQ?usp=sharing
> 
> You can manually download and extract them to `jetson-inference/data/networks`
> I can't use Google drive for the script because it doesn't permit automated downloading (i.e. from `wget`)

I still cannot download the file such as `alexnet.prototxt` and `alexnet_noprob.prototxt`, so when i run `./imagenet-console orange_0.jpg output_0.jpg`, it shows:
`  args (3):  0 [./imagenet-console]  1 [orange_0.jpg]  2 [output_0.jpg]  

imageNet -- loading classification network model from:
         -- prototxt     networks/googlenet.prototxt
         -- model        networks/bvlc_googlenet.caffemodel
         -- class_labels networks/ilsvrc12_synset_words.txt
         -- input_blob   'data'
         -- output_blob  'prob'
         -- batch_size   2

[TRT]  TensorRT version 5.0.6
[TRT]  detected model format - caffe  (extension '.caffemodel')
[TRT]  desired precision specified for GPU: FASTEST
[TRT]  requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]  native precisions detected for GPU:  FP32, FP16
[TRT]  selecting fastest native precision for GPU:  FP16
[TRT]  attempting to open engine cache file networks/bvlc_googlenet.caffemodel.2.1.GPU.FP16.engine
[TRT]  cache file not found, profiling network model on device GPU
[TRT]  device GPU, loading  networks/bvlc_googlenet.caffemodel
[TRT]  CaffeParser: Could not open file 
[TRT]  CaffeParser: Could not parse deploy file
[TRT]  device GPU, failed to parse caffe network
device GPU, failed to load networks/bvlc_googlenet.caffemodel
failed to load networks/bvlc_googlenet.caffemodel
imageNet -- failed to initialize.
imagenet-console:   failed to initialize imageNet`

dusty-nv, 2019-04-08:
You can get Alexnet and Googlenet from here:

https://github.com/BVLC/caffe/tree/master/models/bvlc_alexnet

https://github.com/BVLC/caffe/tree/master/models/bvlc_googlenet

shellyfung, 2019-04-08:
https://github.com/BVLC/caffe/tree/master/models/bvlc_alexnet

https://github.com/BVLC/caffe/tree/master/models/bvlc_googlenet

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/dusty-nv/jetson-inference/issues/72#issuecomment-480802129",
"url": "https://github.com/dusty-nv/jetson-inference/issues/72#issuecomment-480802129",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

rudysetiawan, 2020-03-19:
I think I still had the same issue as well. Sorry for waking this issue up.

Initial, I was not sure what it was but downloading from nvidia.box.com always get stuck at random points. Tried clearing the build directory and started it again, still not ok. Changing to another network provider still produced the same result.
I tested downloading a distro using wget from mirrors.kernel.org, it went just fine.

Then i decided to try to use VPN with a US IP address. I was able to download them all. Not sure it could possibly the CDN that box.com is using for Asia is not working out well?

Mohamed-512, 2020-03-21:
> Hi, here is Google drive link to model folder: https://drive.google.com/drive/folders/0BwYxpotGWRNOV2d1NXNJOHA0dzQ?usp=sharing
>
> You can manually download and extract them to [...]

Any alternatives for PyTorch?
Thanks.

Mohamed-512, 2020-03-21:
I was able to download it  and PyTorch wheel as well using through  https://nvidia.app.box.com/shared/static/{MODEL_ID} using a VPN connected to somewhere in Europe

closed 2023-04-27 by dusty-nv