dusty-nv/jetson-inference #76: Some questions about the deploy precision of jetson.

Seakia, 2017-05-08:
Hi, I used the digits platform to train one classified model and this model achieves one good performance such as the following pic.
[image]

But when I deploy the caffemodel and prototxt file on tx2, the inference results are not good. 

And I do not do anything and I only copy caffemodel, prototxt file, mean.binaryproto and label files and so on which are downloaded from digits and I load these files using the imageNet::Create function.

Do I have some error for the deployment? Do I need to modify somethimgs before using them? Or is it

closed 2017-06-08 by Seakia

sulthanashafi, 2017-06-27:
Have you got it rectified?Iam also having the same problem that jetson not performing that well as if DIGITS do.Do anyone help me,solve it.

Seakia, 2017-06-29:
I didn't solve it. I run caffe library on board. And The FPS of running caffe is close to running tensorRT for my codes.