dusty-nv/jetson-inference #77: Run detectNet for different imagesize without resizing image

csadom, 2017-05-10:
A trained detectnet  have a given imagesize as an input (if the input image is larger or smaller, detectnet resizes them). It is set in the begenning in the deploy.prototxt, as input_shape. If it is set to different size, ot works prorperly too since there is no fully connected layer in the detectnet. However, if it is reset, the network have to rebuild, which is long time, and i would like to run detectnet for different random size in short time. Is there any soluiton to solved this problem? Can I change something in order to not resize image in the begenning od detection?

Thanks

dusty-nv, 2017-05-10:
Hi, instead of resize you could crop. But somehow the input has to match dimension to the network.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

qingzew, 2018-02-04:
@csadom I have the same problem, have you solve it

closed 2023-04-27 by dusty-nv