dusty-nv/jetson-inference #78: Can I run jetson tx2 using FP32?

Seakia, 2017-05-13:
I want to run the imagenet example of tx2 board with FP32 precision. Does the tx2 support the FP32 precision? How can I do it?

dusty-nv, 2017-05-15:
Hi, which vision primitive are you using?  (imageNet, detectNet, or segNet)

Add DisableFP16() call to the constructor code inside imageNet.cpp, detectNet.cpp, or segNet.cpp.  Then it should work.  For example, with imageNet:

```
imageNet::imageNet() : tensorNet()
{
	mOutputClasses = 0;
+       DisableFP16();
}

Seakia, 2017-05-16:
Thanks, I call the DisableFP16() out of constructor so it doesn't work before.

Btw, I also cannot obtain one good classification precision which I got on the PC platform under CPU mode and GPU mode using caffe even if I use disable FP16 to use FP32.

closed 2017-06-08 by Seakia

kptikku, 2018-08-16:
Though adding the 'DisableFP16();' code in imageNet, disabled the FP16(from the logs), the time taken to run the GoogleNet model is the same for both FP16 and FP32. Is that expected, or am I missing something. The logs look identical, even the output probabilities are exactly similar!

I am trying to run this on Nvidia Tegra TX2 running TensorRT 3.0.4