dusty-nv/jetson-inference #83: Fail to load caffemodel of custom detectNet.

YTed, 2017-05-30:
Hi, I had jetson-inference setup on my PC successfully, compilation passed, testings of `detectnet-console` with provided models and images went well.

I trained my own model with KITTI dataset, steps follow with [DIGITS detectNet example](https://github.com/NVIDIA/DIGITS/tree/master/examples/object-detection), the model worked fine on DIGITS, cars could be detected.

Now I want to use this custom detectNet model for jetson-inference. I modified `detectNet.cpp` so that it loads the car model without specifying path. Then I run `detectnet-console`, it prints:

```
$ ./detectnet-console car_0.jpg car_0_detect.jpg -model car
detectnet-console
  args (5):  0 [./detectnet-console]  1 [car_0.jpg]  2 [car_0_detect.jpg]  3 [-model]  4 [car]  

detectNet -- loading detection network model from:
          -- prototxt:   networks/DetectNet-KITTI-Car/deploy.prototxt
          -- model:      networks/DetectNet-KITTI-Car/snapshot_iter_9570.caffemodel
          -- input_blob  'data'
          -- output_cvg  'coverage'
          -- output_bbox 'bboxes'
          -- threshold   0.500000
          -- batch_size  2

[GIE]  attempting to open cache file networks/DetectNet-KITTI-Car/snapshot_iter_9570.caffemodel.2.tensorcache
[GIE]  cache file not found, profiling network model
[GIE]  platform does not have FP16 support.
[GIE]  loading networks/DetectNet-KITTI-Car/deploy.prototxt networks/DetectNet-KITTI-Car/snapshot_iter_9570.caffemodel
conv1/7x7_s2: ERROR - 32-bit weights not found for 32-bit model
conv1/7x7_s2: ERROR - 32-bit weights not found for 32-bit model
[GIE]  Parameter check failed in addConvolution, condition: kernelWeights.values != nullptr
error parsing layer type Convolution index 1
[GIE]  failed to parse caffe network
failed to load networks/DetectNet-KITTI-Car/snapshot_iter_9570.caffemodel
detectNet -- failed to initialize.
detectnet-console:   failed to initialize detectNet
```

Have no idea what to do next. Did I miss something? Any suggestion are welcome. 

Thanks for reading.

dusty-nv, 2017-05-30:
Hmm I am not familiar with this error regarding the 32-bit weights, it also looks like 16-bit is disabled on your platform.  

Is the model corrupted or is there something different about the network you trained?
I wonder if `conv1/7x7_s2` is the first layer it tries to load, or is it the only one with error?

YTed, 2017-05-31:
Hi, @dusty-nv. Thanks for replying.

My GPU is a GTX-1060-6G, with CUDA 8.0 installed. "No FP16 support warning" occurs while testing `DetectNet-COCO-Dog` and other models, either.

The model works on DIGITS(version 5.1-dev). I don't think it's corrupted. I guess there are some parameters I missed while training.

The `deploy.prototxt` is almost identity with `DetectNet-COCO-Dog/deploy.prototxt`, except that input layer is:

```
input_shape {
  dim: 1
  dim: 3
  dim: 384
  dim: 1248
}
```

I wonder how did you train the `DetectNet-COCO-Dog` model. Did you train it with DIGITS? Please let me know.

Thanks again.

dusty-nv, 2017-05-31:
Hi Ted,

The model was trained according to the procedure here:   https://github.com/dusty-nv/jetson-inference#downloading-the-detection-dataset

Are you able to run the pre-trained models that are downloaded with the repo?   For instance:

./detectnet-console dog_1.jpg output_1.jpg coco-dog

Thanks,
Dusty

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

YTed, 2017-05-31:
Hi Dusty,

Yes I can run the pre-trained models. 

Thanks.

YTed, 2017-05-31:
Hi Dusty,

I have re-trained mscoco-dog dataset myself last night. The "32-bit weight" error occurs again while loading `snap_iter_38600.caffemodel`. I think there must be something wrong training the model. I am going to:

1. read relative documents to understand what `.caffemodel` files contain;
2. compare each files of these two models, especially `.caffemodel` files;

Do you have any suggestions for me?

Thanks.

YTed, 2017-06-02:
I dig into the code,and find the error comes from `jetson-inference/tensorNet.cpp`, Line 100:

```
	const nvcaffeparser1::IBlobNameToTensor *blobNameToTensor =
            parser->parse(deployFile.c_str(),		// caffe deploy file
                          modelFile.c_str(),		// caffe model file
                          *network,					// network definition that the parser will populate
                          modelDataType);
```
and `modelDataType` was initialized to `nvinfer1::DataType::kFLOAT`, because of `builder->platformHasFastFp16()` returned false.

I changed the `parser->parse` invocation, hard coded `nvinfer1::DataType::kHALF` as the last parameter, and the `32-bit weight not found` error disappeared, with other error occurs:

```
$ ./detectnet-console dog_0.jpg dog_0_detect.jpg -model car
detectnet-console
  args (5):  0 [./detectnet-console]  1 [dog_0.jpg]  2 [dog_0_detect.jpg]  3 [-model]  4 [car]  

detectNet -- loading detection network model from:
          -- prototxt:   networks/DetectNet-KITTI-Car/deploy.prototxt
          -- model:      networks/DetectNet-KITTI-Car/snapshot_iter_9251.caffemodel
          -- input_blob  'data'
          -- output_cvg  'coverage'
          -- output_bbox 'bboxes'
          -- threshold   0.500000
          -- batch_size  2

[GIE]  attempting to open cache file networks/DetectNet-KITTI-Car/snapshot_iter_9251.caffemodel.2.tensorcache
[GIE]  cache file not found, profiling network model
[GIE]  platform does not have FP16 support.
[GIE]  loading networks/DetectNet-KITTI-Car/deploy.prototxt networks/DetectNet-KITTI-Car/snapshot_iter_9251.caffemodel
[GIE]  retrieved output tensor 'coverage'
[GIE]  retrieved output tensor 'bboxes'
[GIE]  configuring CUDA engine
[GIE]  building CUDA engine
[GIE]  conv1/7x7_s2: kernel weights has count 1 but 9408 was expected
[GIE]  conv2/3x3_reduce: kernel weights has count 1 but 4096 was expected
### many logs like above and below, so I omit them
[GIE]  cvg/classifier: kernel weights has count 1 but 1024 was expected
[GIE]  bbox/regressor: kernel weights has count 1 but 4096 was expected
[GIE]  failed to build CUDA engine
failed to load networks/DetectNet-KITTI-Car/snapshot_iter_9251.caffemodel
detectNet -- failed to initialize.
detectnet-console:   failed to initialize detectNet
```

Just to let you know. 

Besides, I wonder how can my GPU card does not have FP16 supported. I have changed my GPU card to GTX 1080 now! [Wikipeida](https://en.wikipedia.org/wiki/Pascal_(microarchitecture)) said Pascal DOES support FP16, does't it?

omaralvarez, 2017-06-02:
I am also having the ```kernel weights has count 1 but 4096 was expected``` issue with my own trained net. Haven't been able to pinpoint the issue yet. Check out #84.

ga18151, 2017-06-22:
I ran into the same issue.  I think it is a cudnn6 vs cudnn5 issue.  I trained a model on my desktop using cudnn6 and then tested it on my laptop and jetson using GIE and cudnn5 and got the same error 32-bit weight not found.  Have you checked to see that NVcaffe is using cudnn5 on the machine you are using for training?

PiyalGeorge, 2018-10-17:
Can i use this command - 
./detectnet-console dog_0.jpg output_0.jpg \
--prototxt=$NET/deploy.prototxt \
--model=$NET/snapshot_iter_38600.caffemodel \
--input_blob=data \ 
--output_cvg=coverage \
--output_bbox=bboxes

modified with googleNet deploy.prototxt and googleNet.caffemodel? i mean,  what will be the --output_cvg and --output_bbox values for the purpose?

closed 2023-04-27 by dusty-nv