dusty-nv/jetson-inference #84: Can I run a LeNet model on Jetson?

omaralvarez, 2017-05-31:
I have trained my own LeNet model using DIGITS and following the tutorials. I have tested the network on my host node and it works. I now want to run it on my Jetson, but in the code I can only see:

```cpp
enum NetworkType
	{
		ALEXNET,
		GOOGLENET
	};
```

Do I have to extend the class to use LeNet models? Is there a quicker way to run LeNet models?

dusty-nv, 2017-05-31:
Here is the extended usage of imagenet-console that you can use to load custom model without reloading:

```
$ NET=networks/GoogleNet-ILSVRC12-subset

$ ./imagenet-console bird_0.jpg output_0.jpg \
--prototxt=$NET/deploy.prototxt \
--model=$NET/snapshot_iter_184080.caffemodel \
--labels=$NET/labels.txt \
--input_blob=data \
--output_blob=softmax
```

Set the path to your model in the `$NET` variable, change the filenames to match, and set the input/output layers to be the same as your prototxt.

omaralvarez, 2017-05-31:
Thanks! I will test it out!

omaralvarez, 2017-06-01:
I have tested imagenet-console, with the command line parameter modified with my data and first, it only tries to load google-net. After modifying ```imagenet-console.cpp``` to call the correct Create() and after avoiding using ```--*=``` syntax I got it to load my net with the following errors:

```
[GIE]  attempting to open cache file /home/ubuntu/Downloads/snapshot_iter_90.caffemodel.2.tensorcache
[GIE]  cache file not found, profiling network model
[GIE]  platform has FP16 support.
[GIE]  loading /home/ubuntu/Downloads/deploy.prototxt /home/ubuntu/Downloads/snapshot_iter_90.caffemodel
[GIE]  failed to retrieve tensor for output 'prob'
[GIE]  configuring CUDA engine
[GIE]  building CUDA engine
[GIE]  conv1: kernel weights has count 1 but 500 was expected
[GIE]  conv2: kernel weights has count 1 but 25000 was expected
[GIE]  ip1: kernel weights has count 1 but 400000 was expected
[GIE]  ip2: kernel weights has count 1 but 6000 was expected
[GIE]  failed to build CUDA engine
failed to load /home/ubuntu/Downloads/snapshot_iter_90.caffemodel
failed to load /home/ubuntu/Downloads/snapshot_iter_90.caffemodel
imageNet -- failed to initialize.
imagenet-console:   failed to initialize imageNet
```

omaralvarez, 2017-06-01:
Tested the new code, and GetString is not working properly. At least on my Jetson it ignores custom arguments due to the fact that it does not parse the arguments correctly.

omaralvarez, 2017-06-01:
I don't understand the problem. I have tested the caffemodel in the GIE samples and it works. It has the same topology as my custom LeNet model. This points to a problem in the caffemodel which I downloaded from my DIGITS host. 

I have also noticed that mean_path is set to NULL, which I think sometimes will not be correct.

dusty-nv, 2017-06-01:
Something in the model is off with the weights.  See the error messages about having 1 weight in the .caffemodel, while 500 are expected. 

I am finishing up the tutorial section for training ImageNet, it will be interesting if those models work for you.

omaralvarez, 2017-06-01:
Yeah, that would be perfect. Please let me know here and I will try them out. Meanwhile I will install DIGITS in my Jetson and see if the model trained there works. 

Its weird because using DIGITS on the host to classify images it works, I'm thinking it must have something to do with Caffe, maybe the latest Caffe generates incompatible models.

omaralvarez, 2017-06-05:
I think this has something to do with compatibility of the latest NVIDIA Caffe and TensorRT. I just compiled the same Caffe version from the host on the Jetson, and then tested my custom caffemodel with ```tools/caffe``` on the Jetson. Using the Caffe compiled binary it works properly:

```
I0605 07:43:55.638097  7663 caffe.cpp:348] Use GPU with device ID 0
I0605 07:43:56.245698  7663 upgrade_proto.cpp:67] Attempting to upgrade input file specified using deprecated input fields: /home/ubuntu/Downloads/deploy.prototxt
I0605 07:43:56.245824  7663 upgrade_proto.cpp:70] Successfully upgraded file specified using deprecated input fields.
W0605 07:43:56.245872  7663 upgrade_proto.cpp:72] Note that future Caffe releases will only support input layers and not input fields.
I0605 07:43:56.246251  7663 net.cpp:51] Initializing net from parameters: 
state {
  phase: TRAIN
  level: 0
  stage: ""
}
layer {
  name: "input"
  type: "Input"
  top: "data"
  input_param {
    shape {
      dim: 1
      dim: 1
      dim: 28
      dim: 28
    }
  }
}
layer {
  name: "scale"
  type: "Power"
  bottom: "data"
  top: "scaled"
  power_param {
    scale: 0.0125
  }
}
layer {
  name: "conv1"
  type: "Convolution"
  bottom: "scaled"
  top: "conv1"
  param {
    lr_mult: 1
  }
  param {
    lr_mult: 2
  }
  convolution_param {
    num_output: 20
    kernel_size: 5
    stride: 1
    weight_filler {
      type: "xavier"
    }
    bias_filler {
      type: "constant"
    }
  }
}
layer {
  name: "pool1"
  type: "Pooling"
  bottom: "conv1"
  top: "pool1"
  pooling_param {
    pool: MAX
    kernel_size: 2
    stride: 2
  }
}
layer {
[... 225 lines cut ...]
I0605 07:43:57.391326  7663 caffe.cpp:412]      input	backward: 0.0186841 ms.
I0605 07:43:57.391374  7663 caffe.cpp:409]      scale	forward: 0.0923673 ms.
I0605 07:43:57.391422  7663 caffe.cpp:412]      scale	backward: 0.0189663 ms.
I0605 07:43:57.391468  7663 caffe.cpp:409]      conv1	forward: 0.233149 ms.
I0605 07:43:57.391512  7663 caffe.cpp:412]      conv1	backward: 0.72725 ms.
I0605 07:43:57.391557  7663 caffe.cpp:409]      pool1	forward: 0.0776705 ms.
I0605 07:43:57.391603  7663 caffe.cpp:412]      pool1	backward: 0.0193474 ms.
I0605 07:43:57.391654  7663 caffe.cpp:409]      conv2	forward: 0.702601 ms.
I0605 07:43:57.391705  7663 caffe.cpp:412]      conv2	backward: 0.96311 ms.
I0605 07:43:57.391751  7663 caffe.cpp:409]      pool2	forward: 0.0691705 ms.
I0605 07:43:57.391796  7663 caffe.cpp:412]      pool2	backward: 0.0202126 ms.
I0605 07:43:57.391844  7663 caffe.cpp:409]        ip1	forward: 0.320912 ms.
I0605 07:43:57.391890  7663 caffe.cpp:412]        ip1	backward: 0.57225 ms.
I0605 07:43:57.391939  7663 caffe.cpp:409]      relu1	forward: 0.0628695 ms.
I0605 07:43:57.391985  7663 caffe.cpp:412]      relu1	backward: 0.0165358 ms.
I0605 07:43:57.392030  7663 caffe.cpp:409]        ip2	forward: 0.0964052 ms.
I0605 07:43:57.392074  7663 caffe.cpp:412]        ip2	backward: 0.24121 ms.
I0605 07:43:57.392120  7663 caffe.cpp:409]    softmax	forward: 0.0741673 ms.
I0605 07:43:57.392164  7663 caffe.cpp:412]    softmax	backward: 0.0169726 ms.
I0605 07:43:57.392246  7663 caffe.cpp:417] Average Forward pass: 2.41988 ms.
I0605 07:43:57.392283  7663 caffe.cpp:419] Average Backward pass: 3.20697 ms.
I0605 07:43:57.392318  7663 caffe.cpp:421] Average Forward-Backward: 5.85047 ms.
I0605 07:43:57.392360  7663 caffe.cpp:423] Total Time: 292.523 ms.
I0605 07:43:57.392405  7663 caffe.cpp:424] *** Benchmark ends ***
```

dusty-nv, 2017-06-05:
Which NVcaffe version do you have installed on your DIGITS machine?  Can you try 0.15?   I haven't used 0.16

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

omaralvarez, 2017-06-05:
I have 0.16.1 installed in my host. I will try 0.15 ASAP.

Thx!

omaralvarez, 2017-06-06:
Confirmed, branch ```caffe-0.15``` works. It was indeed a compatibility issue.

dusty-nv, 2017-06-06:
Thank you for confirming the issue Omar.  I will note it in the Readme and report it to the team internally.

omaralvarez, 2017-06-06:
I am also wondering if mean substraction is handled in the current code. I have added some code to also accept mean binary files, and I see a ```mMeanPath```, but it looks like it has not been used anywhere. Is this correct? I could help with the implementation of the mean substraction, I was just wondering if I was missing something and it has already been taken into account.

dusty-nv, 2017-06-06:
Pixel mean subtraction is performed in CUDA kernel before TensorRT, I need to fully add the support for image mean subtraction.

omaralvarez, 2017-06-06:
Thanks for the responses, sorry for asking so many questions. The cuda kernel you are referring to is ```cudaPreImageNetMean```? Because if I am not mistaken it substracts a fixed amount that does not use the binary mean file. 

Last question (I promise). My last doubt has to do with batch sizes. Are batch sizes doing anything right now? Because I see that buffers are allocated for the batch size, when the network is loaded. But the classify function uses the ```loadRGBA``` result that only reads a single image. So if one wanted to use higher batch sizes, modifying the aforementioned function to take advantage of the bigger size of the input buffer would be needed. Is this correct?

dusty-nv, 2017-06-06:
No worries, each of the primitives does need updated with a function that takes the batching.  The console and camera programs don't use it currently right now either.  Now that the training guides are complete we can add that stuff as more advanced parts of the tutorial.  BTW here is the ImageNet training guide we were talking about before:  [https://github.com/dusty-nv/jetson-inference#re-training-the-network-with-digits](https://github.com/dusty-nv/jetson-inference#re-training-the-network-with-digits)

omaralvarez, 2017-06-07:
Thanks dusty. I have checked out the new guide and it works properly for me, it is more or less the process I had followed to train my custom net.

Let me know if I can help with anything.

dusty-nv, 2017-06-07:
Thanks Omar, actually the item of mean image subtraction you raised, basically there needs to be a different `cudaPreImageNetMean` preprocessing function which applies the image subtraction as opposed to pixel subtraction - feel free to take a whack at it if you want.

omaralvarez, 2017-06-07:
Sounds good. I will open a new issue, so we don't polute this one. Should I close it since it has been solved?

dusty-nv, 2017-06-07:
Sure, sounds good, thanks.

closed 2017-06-07 by omaralvarez