dusty-nv/jetson-inference #88: RTSP stream GStreamer pipeline not working

omaralvarez, 2017-06-12:
When using the following GStreamer pipeline:

```
rtspsrc location=rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov ! queue ! rtph264depay ! h264parse ! queue ! omxh264dec ! appsink name=mysink
```

I get these errors:

```
imagenet-camera:  camera open for streaming
NvMMLiteOpen : Block : BlockType = 261 
TVMR: NvMMLiteTVMRDecBlockOpen: 7580: NvMMLiteBlockOpen 
NvMMLiteBlockCreate : Block : BlockType = 261 
TVMR: cbBeginSequence: 1166: BeginSequence  1280x720, bVPR = 0, fFrameRate = 0.000000
TVMR: LowCorner Frequency = 100000 
TVMR: cbBeginSequence: 1545: DecodeBuffers = 3, pnvsi->eCodec = 4, codec = 0 
TVMR: cbBeginSequence: 1606: Display Resolution : (1280x720) 
TVMR: cbBeginSequence: 1607: Display Aspect Ratio : (1280x720) 
TVMR: cbBeginSequence: 1649: ColorFormat : 5 
TVMR: cbBeginSequence:1660 ColorSpace = NvColorSpace_YCbCr709
TVMR: cbBeginSequence: 1790: SurfaceLayout = 3
TVMR: cbBeginSequence: 1868: NumOfSurfaces = 7, InteraceStream = 0, InterlaceEnabled = 0, bSecure = 0, MVC = 0 Semiplanar = 1, bReinit = 1, BitDepthForSurface = 8 LumaBitDepth = 8, ChromaBitDepth = 8, ChromaFormat = 5
Allocating new output: 1280x720 (x 9), ThumbnailMode = 0
[gstreamer] gstreamer decoder onPreroll

imagenet-camera:  failed to capture frame
imagenet-camera:  failed to convert from NV12 to RGBA
imageNet::Classify( 0x(nil), 1280, 720 ) -> invalid parameters

imagenet-camera:  failed to capture frame
imagenet-camera:  failed to convert from NV12 to RGBA
imageNet::Classify( 0x(nil), 1280, 720 ) -> invalid parameters
[gstreamer] gstreamer decoder onBuffer
Caps: video/x-raw, format=(string)NV12, width=(int)1280, height=(int)720, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)bt709, framerate=(fraction)0/1
[cuda]  cudaAllocMapped 1382400 bytes, CPU 0x101020000 GPU 0x101020000
[cuda]  cudaAllocMapped 1382400 bytes, CPU 0x101180000 GPU 0x101180000
[cuda]  cudaAllocMapped 1382400 bytes, CPU 0x1012e0000 GPU 0x1012e0000
[cuda]  cudaAllocMapped 1382400 bytes, CPU 0x101440000 GPU 0x101440000
[cuda]  cudaAllocMapped 1382400 bytes, CPU 0x1015a0000 GPU 0x1015a0000
[cuda]  cudaAllocMapped 1382400 bytes, CPU 0x101700000 GPU 0x101700000
[cuda]  cudaAllocMapped 1382400 bytes, CPU 0x101860000 GPU 0x101860000
[cuda]  cudaAllocMapped 1382400 bytes, CPU 0x1019c0000 GPU 0x1019c0000
[cuda]  cudaAllocMapped 1382400 bytes, CPU 0x101b20000 GPU 0x101b20000
[cuda]  cudaAllocMapped 1382400 bytes, CPU 0x101c80000 GPU 0x101c80000
[cuda]  cudaAllocMapped 1382400 bytes, CPU 0x101de0000 GPU 0x101de0000
[cuda]  cudaAllocMapped 1382400 bytes, CPU 0x101f40000 GPU 0x101f40000
[cuda]  cudaAllocMapped 1382400 bytes, CPU 0x1020a0000 GPU 0x1020a0000
[cuda]  cudaAllocMapped 1382400 bytes, CPU 0x102200000 GPU 0x102200000
[cuda]  cudaAllocMapped 1382400 bytes, CPU 0x102360000 GPU 0x102360000
[cuda]  cudaAllocMapped 1382400 bytes, CPU 0x1024c0000 GPU 0x1024c0000
[cuda]   gstreamer camera -- allocated 16 ringbuffers, 1382400 bytes each
[gstreamer] gstreamer changed state from NULL to READY ==> manager
[gstreamer] gstreamer changed state from READY to PAUSED ==> manager
[gstreamer] gstreamer changed state from NULL to READY ==> rtpssrcdemux0
[gstreamer] gstreamer changed state from READY to PAUSED ==> rtpssrcdemux0
[gstreamer] gstreamer changed state from NULL to READY ==> rtpsession0
[gstreamer] gstreamer changed state from READY to PAUSED ==> rtpsession0
[gstreamer] gstreamer msg progress ==> rtspsrc0
[gstreamer] gstreamer changed state from NULL to READY ==> udpsink0
[gstreamer] gstreamer changed state from READY to PAUSED ==> udpsink0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> udpsink0
[gstreamer] gstreamer changed state from NULL to READY ==> fakesrc0
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> fakesrc0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> fakesrc0
[gstreamer] gstreamer msg progress ==> rtspsrc0
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> rtpssrcdemux0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> rtpsession0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> manager
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> udpsrc1
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> udpsrc1
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> udpsrc2
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> udpsrc2
[gstreamer] gstreamer msg progress ==> rtspsrc0
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer changed state from NULL to READY ==> rtpptdemux0
[gstreamer] gstreamer changed state from READY to PAUSED ==> rtpptdemux0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> rtpptdemux0
[gstreamer] gstreamer changed state from NULL to READY ==> rtpjitterbuffer0
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> rtpjitterbuffer0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> rtpjitterbuffer0
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer msg stream-start ==> pipeline0
[gstreamer] gstreamer msg element ==> rtpsession0
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer mysink missing gst_tag_list_to_string()
[cuda]   gstreamer camera -- allocated 16 RGBA ringbuffers
Segmentation fault
```

The segmentation fault happens on the omxdecoder thread, which is really weird. I have tested the pipeline with other code that I made to watch videos on the Jetson and it works properly. I think the error is caused by jetson-inference trying to use the Gstreamer buffer before there is anything in it.

The decoder should output ```video/x-raw, format=(string)NV12``` which is what jetson-inference accepts.

omaralvarez, 2017-06-13:
I think I have fixed it. I will create a pull request to handle rtsp streams when I'm sure it is indeed fixed.

Xuejingyuan, 2017-06-23:
Hello omaralvarez!I'm very sorry to bother you,but I have a question which has not be fixed by myself.I need your help that how can I print log information like yours in my program.How can I write code for gstreamer debug in my program?Thank you firstly.Hope for your reply.

omaralvarez, 2017-06-23:
No problem. There is no need to set ```gst_debug_set_default_threshold(5)```, which in my opinion is too much, anything over 4 is probably too much unless you are a GStreamer developer.

Those messages are generated by jetson-inference not GStreamer, so you should see them without having to do anything extra. Are you using the correct app?

Xuejingyuan, 2017-06-26:
Sorry omaralvarez,I don't understand what your jetson-inference refers to specificially.I used qtcreator.Do you mean NSight?However I think NSight just gives cuda infortmation.

omaralvarez, 2017-06-26:
I will rephrase my answer, all of the executables that this project creates output the information that you mentioned to the command line. Some of them will not output GStreamer info, since they don't use it, like for example: imagenet-console, segnet-console, detectnet-console. Try the ones that end with *-camera, those should generate GStreamer logs.

Xuejingyuan, 2017-06-27:
Now I know.Thank you very much!

turbosnail9, 2017-06-29:
Hi @omaralvarez I compiled your branch and it seems to work. To get the rtsp pipeline did you edit lines 289-290 pf `gstCamera.cpp`?

kanakiyab, 2017-10-20:
@omaralvarez ,

Great work on the RTSP side. I have seen your files from the pull request.

I am trying a similar thing, but for a video file. I am getting a similar error as you did in the beginning. Here it is:

```
nvidia@tegra-ubuntu:~/jetson-inference/build/aarch64/bin$ ./detectnet-camera --prototxt deploy.prototxt --model snapshot_iter_2240.caffemodel
detectnet-camera
  args (5):  0 [./detectnet-camera]  1 [--prototxt]  2 [deploy.prototxt]  3 [--model]  4 [snapshot_iter_2240.caffemodel]  

[gstreamer] initialized gstreamer, version 1.8.3.0
[gstreamer] gstreamer decoder pipeline string:
filesrc location=/home/nvidia/DJI_0021_480.avi ! decodebin ! videoscale ! video/x-raw, format=(string)RGB, width=(int)640, height=(int)360 ! videoconvert ! appsink name=mysink

detectnet-camera:  successfully initialized video device
    width:  640
   height:  360
    depth:  24 (bpp)

detectNet -- loading detection network model from:
          -- prototxt    deploy.prototxt
          -- model       snapshot_iter_2240.caffemodel
          -- input_blob  'data'
          -- output_cvg  'coverage'
          -- output_bbox 'bboxes'
          -- mean_pixel  0.000000
          -- threshold   0.500000
          -- batch_size  2

[GIE]  TensorRT version 2.1, build 2102
[GIE]  attempting to open cache file snapshot_iter_2240.caffemodel.2.tensorcache
[GIE]  loading network profile from cache... snapshot_iter_2240.caffemodel.2.tensorcache
[GIE]  platform has FP16 support.
[GIE]  snapshot_iter_2240.caffemodel loaded
[GIE]  CUDA engine context initialized with 3 bindings
[GIE]  snapshot_iter_2240.caffemodel input  binding index:  0
[GIE]  snapshot_iter_2240.caffemodel input  dims (b=2 c=3 h=360 w=640) size=5529600
[cuda]  cudaAllocMapped 5529600 bytes, CPU 0x102a00000 GPU 0x102a00000
[GIE]  snapshot_iter_2240.caffemodel output 0 coverage  binding index:  1
[GIE]  snapshot_iter_2240.caffemodel output 0 coverage  dims (b=2 c=1 h=45 w=80) size=28800
[cuda]  cudaAllocMapped 28800 bytes, CPU 0x103000000 GPU 0x103000000
[GIE]  snapshot_iter_2240.caffemodel output 1 bboxes  binding index:  2
[GIE]  snapshot_iter_2240.caffemodel output 1 bboxes  dims (b=2 c=4 h=45 w=80) size=115200
[cuda]  cudaAllocMapped 115200 bytes, CPU 0x103200000 GPU 0x103200000
snapshot_iter_2240.caffemodel initialized.
[cuda]  cudaAllocMapped 16 bytes, CPU 0x103400000 GPU 0x103400000
maximum bounding boxes:  14400
[cuda]  cudaAllocMapped 230400 bytes, CPU 0x10321c200 GPU 0x10321c200
[cuda]  cudaAllocMapped 57600 bytes, CPU 0x103007200 GPU 0x103007200
default X screen 0:   1920 x 1080
[OpenGL]  glDisplay display window initialized
[OpenGL]   creating 640x360 texture
loaded image  fontmapA.png  (256 x 512)  2097152 bytes
[cuda]  cudaAllocMapped 2097152 bytes, CPU 0x103600000 GPU 0x103600000
[cuda]  cudaAllocMapped 8192 bytes, CPU 0x103800000 GPU 0x103800000
[gstreamer] gstreamer transitioning pipeline to GST_STATE_PLAYING
[gstreamer] gstreamer changed state from NULL to READY ==> mysink
[gstreamer] gstreamer changed state from NULL to READY ==> videoconvert0
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter0
[gstreamer] gstreamer changed state from NULL to READY ==> videoscale0
[gstreamer] gstreamer changed state from NULL to READY ==> typefind
[gstreamer] gstreamer changed state from NULL to READY ==> decodebin0
[gstreamer] gstreamer changed state from NULL to READY ==> filesrc0
[gstreamer] gstreamer changed state from NULL to READY ==> pipeline0
[gstreamer] gstreamer changed state from READY to PAUSED ==> videoconvert0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter0
[gstreamer] gstreamer changed state from READY to PAUSED ==> videoscale0
[gstreamer] gstreamer stream status CREATE ==> sink
[gstreamer] gstreamer changed state from READY to PAUSED ==> typefind
[... 15 lines cut ...]
[cuda]  cudaAllocMapped 691200 bytes, CPU 0x1042a8c00 GPU 0x1042a8c00
[cuda]  cudaAllocMapped 691200 bytes, CPU 0x104351800 GPU 0x104351800
[cuda]  cudaAllocMapped 691200 bytes, CPU 0x104400000 GPU 0x104400000
[cuda]   gstreamer camera -- allocated 16 ringbuffers, 691200 bytes each
[gstreamer] gstreamer stream status ENTER ==> sink
[gstreamer] gstreamer changed state from NULL to READY ==> avidemux0
[gstreamer] gstreamer stream status CREATE ==> sink
[gstreamer] gstreamer changed state from READY to PAUSED ==> avidemux0
[gstreamer] gstreamer stream status ENTER ==> sink
[gstreamer] gstreamer stream status ENTER ==> src_0
[gstreamer] gstreamer changed state from NULL to READY ==> pngdec0
[gstreamer] gstreamer changed state from READY to PAUSED ==> pngdec0
[gstreamer] gstreamer changed state from READY to PAUSED ==> decodebin0
[gstreamer] gstreamer msg stream-start ==> pipeline0
[gstreamer] gstreamer mysink missing gst_tag_list_to_string()
[gstreamer] gstreamer mysink missing gst_tag_list_to_string()
Segmentation fault (core dumped)
```

My question to you is that how did you figure out that your code failed on the decoder thread? Do you think my pipeline could be failing due to a similar error? I am going to try your code next to see what kind of results/errors I get.

Any pointers will help.

Thanks,
Bhargav

kanakiyab, 2017-10-20:
Thanks, Omar! I am going to try your pull request and see if my problem is solved.

kanakiyab, 2017-10-28:
@omaralvarez , a suggestion for you.

I believe the segfault is due to the assignment of C++ string to a char* (here gchar*). As per one stackoverflow [thread](https://stackoverflow.com/questions/8356223/assign-a-string-literal-to-a-char), I used `strdup()` instead of removing out the lines in [gstUtility.cpp:218](https://github.com/dusty-nv/jetson-inference/blob/058c88a182ee49bcefc83c0b9ac019f7ca449d80/util/camera/gstUtility.cpp#L218), just use `strdup()`.