dusty-nv/jetson-inference #91: unspecified launch failure (error 4) (hex 0x04)

kclim2, 2017-06-14:
Everything is running smoothly until i included the following line into my project:
CUDA_FAILED(cudaRectOutlineOverlay((float4*)rgba, (float4*)rgba, width, height, (float4*)bb, 1, color))

Anyone can suggest what could be the source of error?

[cuda]   cudaDeviceSynchronize()
[cuda]      unspecified launch failure (error 4) (hex 0x04)
[cuda]      /home/ubuntu/jetson-inference/detectNet.cpp:581
[cuda]   cudaGetLastError()
[cuda]      unspecified launch failure (error 4) (hex 0x04)
[cuda]      /home/ubuntu/jetson-inference/util/cuda/cudaNormalize.cu:47
[cuda]   cudaNormalizeRGBA((float4*)imgRGBA, make_float2(0.0f, 255.0f), (float4*)imgRGBA, make_float2(0.0f, 1.0f), camera->GetWidth(), camera->GetHeight())
[cuda]      unspecified launch failure (error 4) (hex 0x04)
[cuda]      /home/ubuntu/jetson-inference/detectnet-camera/detectnet-camera.cpp:247
[cuda]   cudaGraphicsGLRegisterBuffer(&mInteropCUDA, mDMA, cudaGraphicsRegisterFlagsWriteDiscard)
[cuda]      unspecified launch failure (error 4) (hex 0x04)
[cuda]      /home/ubuntu/jetson-inference/util/display/glTexture.cpp:235
[cuda]   cudaGetLastError()
[cuda]      unspecified launch failure (error 4) (hex 0x04)
[cuda]      /home/ubuntu/jetson-inference/util/cuda/cudaYUV-NV12.cu:366
[cuda]   cudaNV12ToRGBAf((uint8_t*)input, (float4*)mRGBA[mLatestRGBA], mWidth, mHeight)
[cuda]      unspecified launch failure (error 4) (hex 0x04)
[cuda]      /home/ubuntu/jetson-inference/util/camera/gstCamera.cpp:84
detectnet-camera:  failed to convert from NV12 to RGBA
detectNet::Detect( 0x(nil), 1280, 720 ) -> invalid parameters
[cuda]   cudaNormalizeRGBA((float4*)imgRGBA, make_float2(0.0f, 255.0f), (float4*)imgRGBA, make_float2(0.0f, 1.0f), camera->GetWidth(), camera->GetHeight())
[cuda]      invalid device pointer (error 17) (hex 0x11)
[cuda]      /home/ubuntu/jetson-inference/detectnet-camera/detectnet-camera.cpp:247
[cuda]   cudaGraphicsGLRegisterBuffer(&mInteropCUDA, mDMA, cudaGraphicsRegisterFlagsWriteDiscard)
[cuda]      unspecified launch failure (error 4) (hex 0x04)
[cuda]      /home/ubuntu/jetson-inference/util/display/glTexture.cpp:235
[cuda]   cudaGetLastError()

closed 2017-06-14 by kclim2

sanersbug, 2017-12-21:
Hello , do you know how to solve the problem ?  I meet the same error when  i run ./detectnet-

baroselfdriving, 2020-02-24:
Hello! I have the same problem using my own model

mayday0208, 2020-11-27:
> 你好！我使用自己的模型时遇到相同的问题

Hi,have you solved the problem?