dusty-nv/jetson-inference #96: segnet camera available?

xiupingmath, 2017-06-22:
hi,could you please offer segnet camera for segmentation on live video?Thanks.

dusty-nv, 2017-06-23:
Thank you, I have checked in `segnet-camera` to the repo.

xiupingmath, 2017-06-26:
Thanks. I have downloaded your code and build it. I use the jetson onboard camera,then I run
  ./segnet-camera  --prototxt=$NET/deploy.prototxt --model=$NET/snapshot_iter_2756640.caffemodel --labels=$NET/cityscapes-labels.txt --colors=$NET/cityscapes-deploy-colors.txt
Error uccured,it output
[GIE]  segNet::Overlay -- s_w 9  s_h 6  s_c 21  s_x 0.018750  s_y 0.016667
[GIE]  segNet::Overlay -- ignoring class 'void' id=0
Segmentation fault (core dumped)

Could you please solve this problem? Thanks.

xiupingmath, 2017-06-26:
Hi, another question, if I want to use my own USB camera, shoule I replace #include "gstCamera.h" in segnet-camera.cpp to #include "v4l2amera.h" ?

phil443, 2017-06-26:
I'm also getting the seg fault running the same command (but with the 367568 model iteration). 

Also by default running segnet-camera attempts to load the model from a "FCN-Alexnet-Cityscapes-SD" directory and with a different model iteration, which both do not exist ("FCN-Alexnet-Cityscapes-HD" does exist however) . Minor issue since the model can be specified manually, but thought I'd point it out.

ShreyasSkandan, 2017-07-05:
I'm experiencing the same problem. Has anyone had any luck with this?

gcjyzdd, 2017-07-10:
I also have the same problem. I found that the program exited with segment faults when I tried to access the memory of imgRGBA(eg,printf("%f\n",((float*)imgRGBA)[0]);) even before the overlay function. Actually, the Overlay function writes the memory of the input (imgRGBA) with segmentation faults. If I comment out the writing part in the Overlay function, the program can run normally and show imgRGBA. I don't know why we cannot access the memory of imgRGBA but it's shown in the image after the function call. 

I also tried to print the memory of imgRGBA in detectnet-camera.cpp and it reported the same error.

dusty-nv, 2017-07-10:
Thanks, I was able to find the problem now, imgRGBA wasn’t allocated with zeroCopy memory.  I patched it and checked in the latest.   If you pull from master it should run now.  Note that the overlay still needs accelerated in CUDA, I will have to get to it sometime or someone else can contribute it.

Best regards,
Dustin

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

ShreyasSkandan, 2017-07-10:
Works for me now! Thanks Dusty

abdallah0012, 2019-05-09:
Hi
 when i run this  ./segnet-camera , i have this error(jetson xavier+logitech c930e)
[TRT]  TensorRT version 5.0.3
[TRT]  detected model format - caffe  (extension '.caffemodel')
[TRT]  desired precision specified for GPU: FASTEST
[TRT]  requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]  native precisions detected for GPU:  FP32, FP16, INT8
[TRT]  selecting fastest native precision for GPU:  FP16
[TRT]  attempting to open engine cache file networks/FCN-Alexnet-Cityscapes-HD/snapshot_iter_367568.caffemodel.2.1.GPU.FP16.engine
[TRT]  cache file not found, profiling network model on device GPU
[TRT]  device GPU, loading networks/FCN-Alexnet-Cityscapes-HD/deploy.prototxt networks/FCN-Alexnet-Cityscapes-HD/snapshot_iter_367568.caffemodel
[TRT]  CaffeParser: Could not parse binary model file
[TRT]  CaffeParser: Could not parse model file
[TRT]  device GPU, failed to parse caffe network
device GPU, failed to load networks/FCN-Alexnet-Cityscapes-HD/snapshot_iter_367568.caffemodel
segNet -- failed to initialize.
segnet-camera:   failed to initialize imageNet

Also, i try with this:
./segnet-camera --prototxt=$NET/deploy.prototxt --model=$NET/snapshot_iter_2756640.caffemodel --labels=$NET/cityscapes-labels.txt --colors=$NET/cityscapes-deploy-colors.txt

[TRT]  TensorRT version 5.0.3
[TRT]  detected model format - caffe  (extension '.caffemodel')
[TRT]  desired precision specified for GPU: FASTEST
[TRT]  requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]  native precisions detected for GPU:  FP32, FP16, INT8
[TRT]  selecting fastest native precision for GPU:  FP16
[TRT]  attempting to open engine cache file .2.1.GPU.FP16.engine
[TRT]  cache file not found, profiling network model on device GPU
[TRT]  device GPU, loading  
[TRT]  CaffeParser: Could not open file 
[TRT]  CaffeParser: Could not parse model file
[TRT]  device GPU, failed to parse caffe network
device GPU, failed to load 
segNet -- failed to initialize.
segnet-camera:   failed to initialize imageNet
 Any help please !!

gaotao19890725, 2020-09-28:
> Works for me now! Thanks Dusty

Hi ShreyasSkandan 
Can you give me some suggestions about how to connect ZED camera with Jetson-inference on TX2 to deal with SegNet problem.
Thanks a million