dusty-nv/jetson-inference #97: some changes been made to TensorRT?

ShervinAr, 2017-06-23:
Hello,

I was previously able to compile all the provided examples in this repo for jetson.
But recently after upgrading my host and reinstalling all required packages (latest updates) I got the following error message while compiling examples for Jetson tx2:

In file included from /home/shervin/Desktop/jetson-inference/segNet.h:9:0,
                 from /home/shervin/Desktop/jetson-inference/segNet.cpp:5:
/home/shervin/Desktop/jetson-inference/tensorNet.h:159:12: error: ‘Dims3’ in namespace ‘nvinfer1’ does not name a type
  nvinfer1::Dims3 mInputDims;
            ^
/home/shervin/Desktop/jetson-inference/tensorNet.h:164:13: error: ‘Dims3’ in namespace ‘nvinfer1’ does not name a type
   nvinfer1::Dims3 dims;
             ^
In file included from /home/shervin/Desktop/jetson-inference/segNet.cpp:5:0:
/home/shervin/Desktop/jetson-inference/segNet.h: In member function ‘uint32_t segNet::GetNumClasses() const’:
/home/shervin/Desktop/jetson-inference/segNet.h:101:66: error: ‘const value_type {aka const struct tensorNet::outputLayer}’ has no member named ‘dims’
  inline uint32_t GetNumClasses() const      { return mOutputs[0].dims.c; }
                                                                  ^
/home/shervin/Desktop/jetson-inference/segNet.cpp: In static member function ‘static segNet* segNet::Create(const char*, const char*, const char*, const char*, const char*, const char*, uint32_t)’:
/home/shervin/Desktop/jetson-inference/segNet.cpp:164:35: error: ‘__gnu_cxx::__alloc_traits<std::allocator<tensorNet::outputLayer> >::value_type {aka struct tensorNet::outputLayer}’ has no member named ‘dims’
  const int s_w = net->mOutputs[0].dims.w;
                                   ^
/home/shervin/Desktop/jetson-inference/segNet.cpp:165:35: error: ‘__gnu_cxx::__alloc_traits<std::allocator<tensorNet::outputLayer> >::value_type {aka struct tensorNet::outputLayer}’ has no member named ‘dims’
  const int s_h = net->mOutputs[0].dims.h;
                                   ^
/home/shervin/Desktop/jetson-inference/segNet.cpp:166:35: error: ‘__gnu_cxx::__alloc_traits<std::allocator<tensorNet::outputLayer> >::value_type {aka struct tensorNet::outputLayer}’ has no member named ‘dims’
  const int s_c = net->mOutputs[0].dims.c;
                                   ^
/home/shervin/Desktop/jetson-inference/segNet.cpp: In member function ‘bool segNet::Overlay(float*, float*, uint32_t, uint32_t, const char*)’:
/home/shervin/Desktop/jetson-inference/segNet.cpp:356:30: error: ‘__gnu_cxx::__alloc_traits<std::allocator<tensorNet::outputLayer> >::value_type {aka struct tensorNet::outputLayer}’ has no member named ‘dims’
  const int s_w = mOutputs[0].dims.w;
                              ^
/home/shervin/Desktop/jetson-inference/segNet.cpp:357:30: error: ‘__gnu_cxx::__alloc_traits<std::allocator<tensorNet::outputLayer> >::value_type {aka struct tensorNet::outputLayer}’ has no member named ‘dims’
  const int s_h = mOutputs[0].dims.h;
                              ^
/home/shervin/Desktop/jetson-inference/segNet.cpp:358:30: error: ‘__gnu_cxx::__alloc_traits<std::allocator<tensorNet::outputLayer> >::value_type {aka struct tensorNet::outputLayer}’ has no member named ‘dims’
  const int s_c = mOutputs[0].dims.c;
                              ^
CMakeFiles/jetson-inference.dir/build.make:1373: recipe for target 'CMakeFiles/jetson-inference.dir/segNet.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/segNet.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/shervin/Desktop/jetson-inference/tensorNet.cpp:5:0:
/home/shervin/Desktop/jetson-inference/tensorNet.h:159:12: error: ‘Dims3’ in namespace ‘nvinfer1’ does not name a type
  nvinfer1::Dims3 mInputDims;
            ^
/home/shervin/Desktop/jetson-inference/tensorNet.h:164:13: error: ‘Dims3’ in namespace ‘nvinfer1’ does not name a type
   nvinfer1::Dims3 dims;
             ^
/home/shervin/Desktop/jetson-inference/tensorNet.cpp: In constructor ‘tensorNet::tensorNet()’:
/home/shervin/Desktop/jetson-inference/tensorNet.cpp:32:10: error: ‘mInputDims’ was not declared in this scope
  memset(&mInputDims, 0, sizeof(nvinfer1::Dims3));
          ^
/home/shervin/Desktop/jetson-inference/tensorNet.cpp:32:32: error: ‘Dims3’ is not a member of ‘nvinfer1’
  memset(&mInputDims, 0, sizeof(nvinfer1::Dims3));
                                ^
/home/shervin/Desktop/jetson-inference/tensorNet.cpp: In member function ‘bool tensorNet::ProfileModel(const string&, const string&, const std::vector<std::__cxx11::basic_string<char> >&, unsigned int, std::ostream&)’:
/home/shervin/Desktop/jetson-inference/tensorNet.cpp:85:58: error: ‘createInferBuilder’ was not declared in this scope
  nvinfer1::IBuilder* builder = createInferBuilder(gLogger);
                                                          ^
/home/shervin/Desktop/jetson-inference/tensorNet.cpp:85:58: note: suggested alternatives:
In file included from /home/shervin/Desktop/jetson-inference/tensorNet.h:9:0,
                 from /home/shervin/Desktop/jetson-inference/tensorNet.cpp:5:
/usr/include/x86_64-linux-gnu/NvInfer.h:2742:18: note:   ‘nvinfer1::{anonymous}::createInferBuilder’
 inline IBuilder* createInferBuilder(ILogger& logger)
                  ^
/usr/include/x86_64-linux-gnu/NvInfer.h:2742:18: note:   ‘nvinfer1::{anonymous}::createInferBuilder’
/home/shervin/Desktop/jetson-inference/tensorNet.cpp:153:34: error: no matching function for call to ‘nvinfer1::ICudaEngine::serialize(std::ostream&)’
  engine->serialize(gieModelStream);
                                  ^
In file included from /home/shervin/Desktop/jetson-inference/tensorNet.h:9:0,
[... 36 lines cut ...]
  nvinfer1::Dims3 inputDims  = engine->getBindingDimensions(inputIndex);
  ^
/home/shervin/Desktop/jetson-inference/tensorNet.cpp:282:37: error: ‘inputDims’ was not declared in this scope
  size_t inputSize  = maxBatchSize * inputDims.c * inputDims.h * inputDims.w * sizeof(float);
                                     ^
/home/shervin/Desktop/jetson-inference/tensorNet.cpp:309:3: error: ‘Dims3’ is not a member of ‘nvinfer1’
   nvinfer1::Dims3 outputDims = engine->getBindingDimensions(outputIndex);
   ^
/home/shervin/Desktop/jetson-inference/tensorNet.cpp:310:38: error: ‘outputDims’ was not declared in this scope
   size_t outputSize = maxBatchSize * outputDims.c * outputDims.h * outputDims.w * sizeof(float);
                                      ^
/home/shervin/Desktop/jetson-inference/tensorNet.cpp:328:5: error: ‘struct tensorNet::outputLayer’ has no member named ‘dims’
   l.dims = outputDims;
     ^
/home/shervin/Desktop/jetson-inference/tensorNet.cpp:335:2: error: ‘mInputDims’ was not declared in this scope
  mInputDims      = inputDims;
  ^
CMakeFiles/jetson-inference.dir/build.make:1397: recipe for target 'CMakeFiles/jetson-inference.dir/tensorNet.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/tensorNet.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/jetson-inference.dir/all' failed
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
Makefile:116: recipe for target 'all' failed
make: *** [all] Error 2

It looks like some variables have changed their names in the latest tensorRT release but the current repo is not updated accordingly?

kyuhyong, 2017-07-25:
Same issue here too
I recently updated my TX2 with newly release Jetpack 3.1 which includes TensorRT 2.1
I found Dims3 changed to DimsCHW
Other issues are relate to calling functions with different name or different params which I am looking for.
Looks like many changes has been made to tensorrt 2.1 from 1.0

dusty-nv, 2017-07-26:
Hi guys, support for TensorRT2 is working now in master with commit [e40bd6](https://github.com/dusty-nv/jetson-inference/commit/e40bd6abe9f07992f65dffeaa19e4d32eef78778)

For backwards compatibility, it's still building for previous JetPack's still on TensorRT1 using some macros I added. There were no outward-facing changes to the vision primitive APIs themselves.

closed 2017-07-26 by dusty-nv