dusty-nv/jetson-inference #99: related to issue #84

sansoy, 2017-06-25:
On a single class trained detectnet i am able to run detectnet-camera successfully on my Jetson TX1 with a webcam.

I also successfully trained on a multiple of 3 classes on my DIGITS box and the tests ran fine there.

However, when i port this multiclass model to my Jetson TX1 I get the error thats popping up in issue #84, ie "kernel weights has count 1 but 12288 was expected"

and i confirmed the DIGITS box caffe version is 0.15

so not sure whats going on.  

I did have to customize the prototxt file on the DIGITS side to accommodate the extra classes and the tests ran fine.

full error message listed below.

ubuntu@tegra-ubuntu:~/jetson-inference/build/aarch64/bin$ ./detectnet-camera beachtrash
detectnet-camera
  args (2):  0 [./detectnet-camera]  1 [beachtrash]  

[gstreamer] initialized gstreamer, version 1.8.3.0
[gstreamer] gstreamer decoder pipeline string:
v4l2src device=/dev/video0 ! video/x-raw, width=(int)1280, height=(int)720, format=RGB ! videoconvert ! video/x-raw, format=RGB ! videoconvert !appsink name=mysink

detectnet-camera:  successfully initialized video device
    width:  1280
   height:  720
    depth:  24 (bpp)

detectNet -- loading detection network model from:
          -- prototxt    networks/beachtrash/deploy.prototxt
          -- model       networks/beachtrash/snapshot_iter_2820.caffemodel
          -- input_blob  'data'
          -- output_cvg  'coverage'
          -- output_bbox 'bboxes'
          -- threshold   0.500000
          -- batch_size  2

[GIE]  attempting to open cache file networks/beachtrash/snapshot_iter_2820.caffemodel.2.tensorcache
[GIE]  cache file not found, profiling network model
[GIE]  platform has FP16 support.
[GIE]  loading networks/beachtrash/deploy.prototxt networks/beachtrash/snapshot_iter_2820.caffemodel
[GIE]  retrieved output tensor 'coverage'
[GIE]  retrieved output tensor 'bboxes'
[GIE]  configuring CUDA engine
[GIE]  building CUDA engine
[GIE]  conv1/7x7_s2: kernel weights has count 1 but 9408 was expected
[GIE]  conv2/3x3_reduce: kernel weights has count 1 but 4096 was expected
[GIE]  conv2/3x3: kernel weights has count 1 but 110592 was expected
[GIE]  inception_3a/1x1: kernel weights has count 1 but 12288 was expected
[GIE]  inception_3a/3x3_reduce: kernel weights has count 1 but 18432 was expected
[GIE]  inception_3a/3x3: kernel weights has count 1 but 110592 was expected
[GIE]  inception_3a/5x5_reduce: kernel weights has count 1 but 3072 was expected
[GIE]  inception_3a/5x5: kernel weights has count 1 but 12800 was expected
[GIE]  inception_3a/pool_proj: kernel weights has count 1 but 6144 was expected
[GIE]  inception_3b/1x1: kernel weights has count 1 but 32768 was expected
[GIE]  inception_3b/3x3_reduce: kernel weights has count 1 but 32768 was expected
[GIE]  inception_3b/3x3: kernel weights has count 1 but 221184 was expected
[GIE]  inception_3b/5x5_reduce: kernel weights has count 1 but 8192 was expected
[GIE]  inception_3b/5x5: kernel weights has count 1 but 76800 was expected
[GIE]  inception_3b/pool_proj: kernel weights has count 1 but 16384 was expected
[GIE]  inception_4a/1x1: kernel weights has count 1 but 92160 was expected
[GIE]  inception_4a/3x3_reduce: kernel weights has count 1 but 46080 was expected
[GIE]  inception_4a/3x3: kernel weights has count 1 but 179712 was expected
[GIE]  inception_4a/5x5_reduce: kernel weights has count 1 but 7680 was expected
[GIE]  inception_4a/5x5: kernel weights has count 1 but 19200 was expected
[GIE]  inception_4a/pool_proj: kernel weights has count 1 but 30720 was expected
[GIE]  inception_4b/1x1: kernel weights has count 1 but 81920 was expected
[GIE]  inception_4b/3x3_reduce: kernel weights has count 1 but 57344 was expected
[GIE]  inception_4b/3x3: kernel weights has count 1 but 225792 was expected
[GIE]  inception_4b/5x5_reduce: kernel weights has count 1 but 12288 was expected
[GIE]  inception_4b/5x5: kernel weights has count 1 but 38400 was expected
[GIE]  inception_4b/pool_proj: kernel weights has count 1 but 32768 was expected
[GIE]  inception_4c/1x1: kernel weights has count 1 but 65536 was expected
[GIE]  inception_4c/3x3_reduce: kernel weights has count 1 but 65536 was expected
[GIE]  inception_4c/3x3: kernel weights has count 1 but 294912 was expected
[GIE]  inception_4c/5x5_reduce: kernel weights has count 1 but 12288 was expected
[GIE]  inception_4c/5x5: kernel weights has count 1 but 38400 was expected
[GIE]  inception_4c/pool_proj: kernel weights has count 1 but 32768 was expected
[GIE]  inception_4d/1x1: kernel weights has count 1 but 57344 was expected
[GIE]  inception_4d/3x3_reduce: kernel weights has count 1 but 73728 was expected
[GIE]  inception_4d/3x3: kernel weights has count 1 but 373248 was expected
[GIE]  inception_4d/5x5_reduce: kernel weights has count 1 but 16384 was expected
[GIE]  inception_4d/5x5: kernel weights has count 1 but 51200 was expected
[GIE]  inception_4d/pool_proj: kernel weights has count 1 but 32768 was expected
[GIE]  inception_4e/1x1: kernel weights has count 1 but 135168 was expected
[GIE]  inception_4e/3x3_reduce: kernel weights has count 1 but 84480 was expected
[GIE]  inception_4e/3x3: kernel weights has count 1 but 460800 was expected
[GIE]  inception_4e/5x5_reduce: kernel weights has count 1 but 16896 was expected
[GIE]  inception_4e/5x5: kernel weights has count 1 but 102400 was expected
[GIE]  inception_4e/pool_proj: kernel weights has count 1 but 67584 was expected
[GIE]  inception_5a/1x1: kernel weights has count 1 but 212992 was expected
[GIE]  inception_5a/3x3_reduce: kernel weights has count 1 but 133120 was expected
[GIE]  inception_5a/3x3: kernel weights has count 1 but 460800 was expected
[GIE]  inception_5a/5x5_reduce: kernel weights has count 1 but 26624 was expected
[GIE]  inception_5a/5x5: kernel weights has count 1 but 102400 was expected
[GIE]  inception_5a/pool_proj: kernel weights has count 1 but 106496 was expected
[GIE]  inception_5b/1x1: kernel weights has count 1 but 319488 was expected
[GIE]  inception_5b/3x3_reduce: kernel weights has count 1 but 159744 was expected
[GIE]  inception_5b/3x3: kernel weights has count 1 but 663552 was expected
[GIE]  inception_5b/5x5_reduce: kernel weights has count 1 but 39936 was expected
[GIE]  inception_5b/5x5: kernel weights has count 1 but 153600 was expected
[GIE]  inception_5b/pool_proj: kernel weights has count 1 but 106496 was expected
[GIE]  cvg/classifier: kernel weights has count 1 but 3072 was expected
[GIE]  bbox/regressor: kernel weights has count 1 but 4096 was expected
[GIE]  failed to build CUDA engine
failed to load networks/beachtrash/snapshot_iter_2820.caffemodel
detectNet -- failed to initialize.
detectnet-camera:   failed to initialize imageNet
ubuntu@tegra-ubuntu:~/jetson-inference/build/aarch64/bin$

sansoy, 2017-06-26:
my bad.  caffe version is .16.   so downgrading to .15

closed 2017-06-26 by sansoy

sani1486, 2018-03-15:
did it work after downgaading to .15 ?