dusty-nv/jetson-inference #103: segmentation overlay scaling question

ShreyasSkandan, 2017-07-07:
So I tried to do an inference run with FCN-AlexNet-Cityscapes-HD using the pre-trained model that's included in this repository.

When using the test image (peds-001.jpg - 1920x1080), I get pretty reasonable results:
[image]

However, when I scale down the image to roughly 640x480, the results look wrong. Since this is an FCN network, the scale of the input image shouldn't have such a drastic effect on the pixel-wise predictions (right?),
[image]

Any thoughts?
Thanks in advance,

Shreyas

dusty-nv, 2017-07-07:
Hi Shreyas,

The segmentation network need be retrained to accept different resolutions.  The segNet code in the repo simply scales the video down to the appropriate res for the network.  It isn't recommended to scale up, instead use the SD or smaller model.

Best regards,
Dustin

[output4]<https://user-images.githubusercontent.com/5595336/27963266-771b534e-6302-11e7-95a5-55a98b44e806.jpg>

[output5]<https://user-images.githubusercontent.com/5595336/27963356-d4c7750e-6302-11e7-8808-b1ad18781f35.jpg>

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

ShreyasSkandan, 2017-07-07:
Thanks for clearing that up and for the quick response!

closed 2017-07-07 by ShreyasSkandan