dusty-nv/jetson-inference #106: RPROIFused(INvPlugin2) cannot connect to Convolution Layer

airxcy, 2017-07-13:
When Trying my faster-rcnn model (based on resnet ) on the sampleFasterRCNN.cpp in the TensorRT2.1.2 samples. I've encountered a dimensionality issue between RPROIFused(INvPlugin2) layer and Convolution layer during Caffe parser. 
Below is the section of the faster-rcnn where the error occurs. 
-----------------------------------------------
...
layer {
  name: "RPROIFused"
  type: "IPlugin"
  bottom: 'rpn_cls_prob_reshape'
  bottom: 'rpn_bbox_pred'
  bottom: 'res4'
  bottom: 'im_info'
  top: 'rois'
  top: 'pool5'
  top: 'count'
}

layer {
  name: "res5a_1"
  type: "Convolution"
  bottom: "pool5"
  top: "res5a_1"
  param {
    lr_mult: 1
    decay_mult: 1
  }
  convolution_param {
    num_output: 256
    bias_term: false
    pad: 1
    kernel_size: 3
    stride: 2
    weight_filler {
      type: "xavier"
    }
    bias_filler {
      type: "constant"
      value: 0
    }
  }
}
...
------------------------------------------------------
the overall network batch size is set to 1. the output size of the RPROIFused(INvPlugin2) layer is [10,256,7,7], the number of proposal is 10, thus batch size after roi pooling is 10, which is different from the network batch size. thus I've encountered the error during caffe parser:
------------------------------------
sample_fasterRCNN: helpers.cpp:45: nvinfer1::DimsCHW nvinfer1::toCHW(const nvinfer1::Dims&): Assertion `d.nbDims == 3 && d.type[0] == DimensionType::kCHANNEL && d.type[1] == DimensionType::kSPATIAL && d.type[2] == DimensionType::kSPATIAL' failed.
-------------------------------------
I've try another faster-rcnn based on zfmodel, which connect the RPROIFused(INvPlugin2) layer with fully connected layer, and works as intended. So, Is there any future plan to support the resnet based faster-rcnn? and, Is there any work-around for this problem?

Thanks!

zimenglan-sysu-512, 2017-08-04:
hi @airxcy do you know how to modify `RPROIFused(INvPlugin2) layer`?
thanks.

airxcy, 2017-08-04:
@zimenglan-sysu-512 It is not open source, so you cannot change it. But you can override the class and some of its member functions.

zimenglan-sysu-512, 2017-08-04:
hi @airxcy, 

got u, but how can i have a look at this layer so that i can override the class?

so thanks.

airxcy, 2017-08-04:
@zimenglan-sysu-512 in the NvInferPlugin.h header file, It defines "class INvPlugin : public IPlugin", I override it to "INvPlugin2" to print some messages. So there is Only  INvPlugin specified for FasterRCNN. However, you cannot modify the mechanism of this layer because it is highly optimized and compiled into binary. But you can implement ROI+RP layer yourself through IPlugin Interface.

zimenglan-sysu-512, 2017-08-04:
hi @airxcy,

though i look into `jetson-inference` and `nvcaffe`, i have no idea where `NvInferPlugin.h` and `NvCaffeParser.h` are. I just wonder how do you find them?

Thanks.

airxcy, 2017-08-04:
@zimenglan-sysu-512 It's part of TensorRT 2.1.2, if you successfully installed TensorRT 2.1.2. you can use "locate" command to find it. It's usually in /usr/include/ or /usr/include/aarch64-linux-gnu/NvInferPlugin.h for TX1/2.

zimenglan-sysu-512, 2017-08-04:
hi @airxcy，

you are right, it can't be modified, since it only provide the .h file and is compiled into binary. but i wonder, how it run the model, is it running nvcaffe?

btw, do you have tried encapsulating tensorrt into bvlc/caffe as one layer?

thanks.

airxcy, 2017-08-04:
@zimenglan-sysu-512 It's only using tensorRT in the runtime. tensorRT is a fast inference Optimizer, It would make more sense to encapsulate bvlc/caffe into tensorRT, not the other way around. you can use the "Iplugin" class to create your bvlc/caffe layer for tensorRT.

dusty-nv, 2017-09-26:
If you have further question about TensorRT fasterRCNN plugin, please post a question to the Jetson [DevTalk](https://devtalk.nvidia.com/default/board/188/jetson-tx2/) forum or see AastaNV's [plugin sample](https://github.com/AastaNV/Face-Recognition).

closed 2017-09-26 by dusty-nv

haozhaopop, 2017-10-17:
I am facing the exactly same problem, @airxcy , did you find out any solution about this?

airxcy, 2017-10-17:
split the net in two. after RPROIfuse. Foward 2 net use different batch size

airxcy, 2017-10-17:
@haozhaopop split the net in two. after RPROIfuse. Foward 2 net use different batch size

airxcy, 2017-10-17:
@haozhaopop Split the net in two after the RPROIfuse layer. then foward 2 net use different batch size.

haozhaopop, 2017-10-19:
@airxcy , really thanks for your reply, boss, I am just doing the exactly same thing as you told, did you success? I tried to apply caffeToGIEModel in samplefasterrcnn.cpp two times to read my model half by half, but then the function parses second half, error comes out because of segmentation fault, did you ever met same problem?

airxcy, 2017-10-24:
@haozhaopop I succeeded. I've encountered several segmentation faults and fixed them. There can be many reasons, you have to modify every aspect of that code to make it work. You can look into: Memory Buffer, reshape plugin, workspace size, plugin factory, model parameter, separate 2 engine objects. etc

code-command, 2018-01-07:
@airxcy  Hello, you can explain in detail how you are modifying the code? I also encountered the same problem.