dusty-nv/jetson-inference #107: Get different results on jetson with DIGITS on detectnet inference

pooluut, 2017-07-13:
I'm using the detectnet on Jetson TX2.

I trained my models on DIGITS on PC
the inference result is:
[image]

But I deploy my model to tx2, I got a different result on the same input image:
[image]

I also modified the threshold of coverage, it doesn't work.
Any  ideas?

Fistcar, 2017-07-27:
Did you ever find a solution? I am having a similar problem. Using digits I get the right answer for a multi-class detectnet, on TX2 I get a different wrong answer.

dusty-nv, 2017-09-26:
Hi, does your network have a layer with mean value subtraction?  The DetectNet code performs mean subtraction.  You probably need to compute the mean pixel value and pass it to [`DetectNet::Create`](https://github.com/dusty-nv/jetson-inference/blob/058c88a182ee49bcefc83c0b9ac019f7ca449d80/detectNet.h#L106) function.  

Right now the default mean pixel value is setup for ImageNet, however I bet your license plate database may have a significantly different mean value.

closed 2017-09-26 by dusty-nv