dusty-nv/jetson-inference #111: About Int8

jack55436001, 2017-07-27:
Hello I want to knew if Tx2 support Int8
and how to modify the code to support Int8 on Tx2
I try following code 
	builder->setInt8Mode(true);
	nvinfer1::IInt8Calibrator* calibrator;
	builder->setInt8Calibrator(calibrator);
But got error
Did everyone have some hints?

dusty-nv, 2017-07-27:
Hi, TX2 doesn’t support INT8.  Xavier has been announced to include it, though.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

jack55436001, 2017-07-27:
Thanks a lot!!

closed 2017-07-27 by jack55436001