dusty-nv/jetson-inference #113: Build From Source On Jetson (TX2) Fails With JetPack 3.1

wfs, 2017-07-28:
In section "Configuring With Make" in https://github.com/dusty-nv/jetson-inference#building-from-source-on-jetson

I get the following errors ...

nvidia@tegra-ubuntu:~/jetson-inference/build$ cmake ../
CUDA_TOOLKIT_ROOT_DIR not found or specified
-- Could NOT find CUDA (missing:  CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) 
-- system arch:  aarch64
-- output path:  /home/nvidia/jetson-inference/build/aarch64
-- Copying /home/nvidia/jetson-inference/segNet.h
-- Copying /home/nvidia/jetson-inference/detectNet.h
-- Copying /home/nvidia/jetson-inference/tensorNet.h
-- Copying /home/nvidia/jetson-inference/imageNet.h
-- Copying /home/nvidia/jetson-inference/util/commandLine.h
-- Copying /home/nvidia/jetson-inference/util/loadImage.h
-- Copying /home/nvidia/jetson-inference/util/camera/gstCamera.h
-- Copying /home/nvidia/jetson-inference/util/camera/v4l2Camera.h
-- Copying /home/nvidia/jetson-inference/util/camera/gstUtility.h
-- Copying /home/nvidia/jetson-inference/util/cuda/cudaResize.h
-- Copying /home/nvidia/jetson-inference/util/cuda/cudaOverlay.h
-- Copying /home/nvidia/jetson-inference/util/cuda/cudaNormalize.h
-- Copying /home/nvidia/jetson-inference/util/cuda/cudaMappedMemory.h
-- Copying /home/nvidia/jetson-inference/util/cuda/cudaRGB.h
-- Copying /home/nvidia/jetson-inference/util/cuda/cudaFont.h
-- Copying /home/nvidia/jetson-inference/util/cuda/cudaUtility.h
-- Copying /home/nvidia/jetson-inference/util/cuda/cudaYUV.h
-- Copying /home/nvidia/jetson-inference/util/display/glUtility.h
-- Copying /home/nvidia/jetson-inference/util/display/glDisplay.h
-- Copying /home/nvidia/jetson-inference/util/display/glTexture.h
-- Copying /home/nvidia/jetson-inference/data/images/orange_0.jpg
-- Copying /home/nvidia/jetson-inference/data/images/drone_0427.png
-- Copying /home/nvidia/jetson-inference/data/images/dog_2.jpg
-- Copying /home/nvidia/jetson-inference/data/images/red_apple_0.jpg
-- Copying /home/nvidia/jetson-inference/data/images/drone_0428.png
-- Copying /home/nvidia/jetson-inference/data/images/peds-001.jpg
-- Copying /home/nvidia/jetson-inference/data/images/vehicle_0.jpg
-- Copying /home/nvidia/jetson-inference/data/images/peds-002.jpg
-- Copying /home/nvidia/jetson-inference/data/images/drone_0255.png
-- Copying /home/nvidia/jetson-inference/data/images/vehicle_1.jpg
-- Copying /home/nvidia/jetson-inference/data/images/peds-006.png
-- Copying /home/nvidia/jetson-inference/data/images/bird_0.jpg
-- Copying /home/nvidia/jetson-inference/data/images/vehicle_3.jpg
-- Copying /home/nvidia/jetson-inference/data/images/peds-008.png
-- Copying /home/nvidia/jetson-inference/data/images/drone_0436.png
-- Copying /home/nvidia/jetson-inference/data/images/banana_0.jpg
-- Copying /home/nvidia/jetson-inference/data/images/dog_0.jpg
-- Copying /home/nvidia/jetson-inference/data/images/drone_0435.png
-- Copying /home/nvidia/jetson-inference/data/images/fontmapA.png
-- Copying /home/nvidia/jetson-inference/data/images/cat_0.jpg
-- Copying /home/nvidia/jetson-inference/data/images/vehicle_2.jpg
-- Copying /home/nvidia/jetson-inference/data/images/bottle_0.jpg
-- Copying /home/nvidia/jetson-inference/data/images/airplane_0.jpg
-- Copying /home/nvidia/jetson-inference/data/images/granny_smith_0.jpg
-- Copying /home/nvidia/jetson-inference/data/images/peds-007.png
-- Copying /home/nvidia/jetson-inference/data/images/peds-004.png
-- Copying /home/nvidia/jetson-inference/data/images/orange_1.jpg
-- Copying /home/nvidia/jetson-inference/data/images/fontmapB.png
-- Copying /home/nvidia/jetson-inference/data/images/peds-005.png
-- Copying /home/nvidia/jetson-inference/data/images/dog_1.jpg
-- Copying /home/nvidia/jetson-inference/data/images/peds-003.jpg
-- Copying /home/nvidia/jetson-inference/data/images/granny_smith_1.jpg
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.11") 
-- Doxygen detected, to build API reference, run `make docs`
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_CUDART_LIBRARY (ADVANCED)
    linked by target "jetson-inference" in directory /home/nvidia/jetson-inference
    linked by target "imagenet-console" in directory /home/nvidia/jetson-inference/imagenet-console
    linked by target "imagenet-camera" in directory /home/nvidia/jetson-inference/imagenet-camera
    linked by target "detectnet-console" in directory /home/nvidia/jetson-inference/detectnet-console
[... 102 lines cut ...]
   used as include directory in directory /home/nvidia/jetson-inference/util/camera/v4l2-console
   used as include directory in directory /home/nvidia/jetson-inference/util/camera/v4l2-console
   used as include directory in directory /home/nvidia/jetson-inference/util/camera/v4l2-console
   used as include directory in directory /home/nvidia/jetson-inference/util/camera/v4l2-console
   used as include directory in directory /home/nvidia/jetson-inference/util/camera/v4l2-console
   used as include directory in directory /home/nvidia/jetson-inference/util/camera/v4l2-console
   used as include directory in directory /home/nvidia/jetson-inference/util/camera/v4l2-console
   used as include directory in directory /home/nvidia/jetson-inference/util/camera/v4l2-display
   used as include directory in directory /home/nvidia/jetson-inference/util/camera/v4l2-display
   used as include directory in directory /home/nvidia/jetson-inference/util/camera/v4l2-display
   used as include directory in directory /home/nvidia/jetson-inference/util/camera/v4l2-display
   used as include directory in directory /home/nvidia/jetson-inference/util/camera/v4l2-display
   used as include directory in directory /home/nvidia/jetson-inference/util/camera/v4l2-display
   used as include directory in directory /home/nvidia/jetson-inference/util/camera/v4l2-display
   used as include directory in directory /home/nvidia/jetson-inference/docs
   used as include directory in directory /home/nvidia/jetson-inference/docs
   used as include directory in directory /home/nvidia/jetson-inference/docs
   used as include directory in directory /home/nvidia/jetson-inference/docs
   used as include directory in directory /home/nvidia/jetson-inference/docs
   used as include directory in directory /home/nvidia/jetson-inference/docs
   used as include directory in directory /home/nvidia/jetson-inference/docs

-- Configuring incomplete, errors occurred!
See also "/home/nvidia/jetson-inference/build/CMakeFiles/CMakeOutput.log".
See also "/home/nvidia/jetson-inference/build/CMakeFiles/CMakeError.log".

oriondream, 2017-08-08:
You don't have CUDA Toolkit properly installed.

dusty-nv, 2017-09-26:
When flashing your TX2 with JetPack 3.1, please select to install CUDA Toolkit, cuDNN, and TensorRT to the target and complete the post-install steps (over ethernet).   Then CUDA should be installed correctly after rebooting.

closed 2017-09-26 by dusty-nv