dusty-nv/jetson-inference #116: fcn-alexnet-pascal-voc

S4WRXTTCS, 2017-07-31:
I'm not getting the expected results when I run the following command

./segnet-console vehicle_1.jpg test1.png fcn-alexnet-pascal-voc

Here is what the output looks like.

[image]

S4WRXTTCS, 2017-07-31:
I did try images directly from the VOC database, and got similar results.

The fcn-alexnet-aerial-fpv-720p model works correctly. So there doesn't seem to be anything wrong with the install.

I'm using the latest commit on July 26th (bed5c46) with Jetpack 3.1

dusty-nv, 2017-08-11:
Hi S4WRXTTCS, sorry for the delay.  I am not sure what's up with Pascal-VOC model, if it's a change with JetPack 3.1 or with my bilinear interpolation of the overlay in segNet.cpp.  Those lower-resolution vehicle images are tested with the [googlenet-12](https://github.com/dusty-nv/jetson-inference#customizing-the-object-classes) step of the tutorial (image recognition).  For vehicle segmentation, I would try the higher-resolution models like Cityscapes.

I've responded with more detail on the DevTalk thread here:   
https://devtalk.nvidia.com/default/topic/1021784/jetson-tx2/unable-to-get-segmentation-to-work-with-jetson-tx2/post/5201703/#5201703

S4WRXTTCS, 2017-08-11:
I don't believe anything changed with Jetpack 3.1

I had similar issues with it back when it was first added, but that was very early on when you first added it. This was the results back then.

https://github.com/dusty-nv/jetson-inference/issues/67

Austriker, 2017-08-17:
I also have the same issue running tensorRT on a x86 computer with a GTX1070.
I used FCN-Alexnet with a custom dataset with high resolution pictures 2000x2000. On Digits the inference works properly. But with tensorRT i get the same king of results than @S4WRXTTCS .
For now I use caffe directly.

closed 2017-09-26 by dusty-nv